keyboardio/Chrysalis #1234: Shifted keys type unshifted variants about 10% of the time

ianthehenry, 2022-11-17:
**Describe the bug**

I have dedicated punctuation keys to type characters like `!` and `(` without having to press shift. Very often when I press these keys, they instead type the unshifted variants (`1` and `9`).

**To Reproduce**

Steps to reproduce the behavior:

1. Make a key binding for `!`.
2. Press it several times over an RDP connection.
3. Notice that about 10% of the time it types `1` instead of `!`.

**Expected behavior**

It should always send the key that it's supposed to send.

**Desktop (please complete the following information):**

 - OS: macOS
 - Chrysalis Version: 0.12.0

**Additional context**

I am working over RDP, and I worry that this might be a factor. I'd like to adjust the delay in between the shift key events and the other keys events, but I can't find a way to do this.

obra, 2022-11-17:
Hi Ian,

This is a long-standing known bug in RDP that can affect any kind of
"non-standard" keyboard config. Depending on the exact version of RDP,
there may be config options on the app's side that improve keyboard
handling.

obra, 2024-11-11:
Closing as it's an interaction with RDP and any kind of non-standard keyboard and not at all something chrysalis specific.

closed 2024-11-11 by obra