keyboardio/Chrysalis #1239: Install documentation

khers, 2022-12-11:
**Is your feature request related to a problem? Please describe.**
I am a new Chrysalis user and it is not clear to me how to install on a Debian based distro.

**Describe the solution you'd like**
A page that describes the install process from package (or source).

Rom1deTroyes, 2022-12-12:
I don't think there is any unusual thing to do, and it works basically the same way on every system :

- Download the appimage (GNU/Lunix), exe (MSWindows) or dmg (MacOS) file from the [Releases](https://github.com/keyboardio/Chrysalis/releases) page
- Run the appimage as any other executable.

For GNU/Linux, you may have to make it executable first (`chmod u+x` or right cliking depending of your settings)

As you think the documentation is not clear on this point, maybe you can formulate this in a way that seems more understandable (as an old console user, I'm not good at user-friendly desktop explanations :-$), so we can improve this part ?

korbinian90, 2024-01-26:
I had the problem that I didn't know appimage is used in Linux and downloaded the *linux*.zip file. This explanation helped!

Maybe adjust the documentation to:

### From installer packages

Choose whether to install the production release, or a development snapshot from the links [above](#downloads), and from the release page, select the download appropriate for your operating system: appimage (GNU/Linux), exe (MSWindows) or dmg (MacOS).

obra, 2024-02-26:
Chrysalis now runs in-browser using WebSerial and WebUSB, so I'm closing out this issue as obsolete. Please don't hesitate to open a new issue if https://chrysalis.keyboard.io exhibits the same behavior

closed 2024-02-26 by obra