keyboardio/Chrysalis #1241: Corruption and chaos

HeavyThumper, 2022-12-21:
About once a week or so, apparently my Model 100/Chrysalis decide my layout is just too boring. They want to make things exciting. So they spontaneously change my keys, change my layers, and add layers. Fun.

What's even more joyous is it seems to affect my backups as well. I haven't delved extensively into this yet but now that it's about the 4th time I think I need to share this. I'm going to backup, copy the backup file just in case, test a restore - which all works. Then maybe in another week things will shake up again - maybe this time my backup backup will restore better.

obra, 2022-12-21:
That's no good. Can you tell us a bit more about your OS and hardware
setup? Also, can you confirm what version of Chrysalis and the Model 100
firmware you're running?

Does this happen...spontaneously or while you're making changes in
Chrysalis?

HeavyThumper, 2022-12-23:
Spontaneously. Windows 11, keyboard plugged directly to MB USB port.

Now I've got a new one - my right side keyboard colors are totally different from the program. Layer 0 is fine. As is Layer 1 - but Layer 2 (my numpad setup) most of the switches don't change color and if they do they certainly don't match the definition. And now the numpad key is fading in/out blue - I don't even know how to program that.

obra, 2022-12-23:
That’s just the number pad mode. Tapping the num key should toggle it on and off.

HeavyThumper, 2022-12-23:
So the numlock key, which I have in Chrysalis as LockTo#1/Yellow, LockTo#0/Red, and F11/Cyan - is now a pulsing blue regardless of the layer. My layer 0/2 works (I misspoke earlier - layer 1 is my numpad layer) but layer 1 while it seems to be at least somewhat functional the colors either don't change or change to something else.

HeavyThumper, 2023-03-04:
I don't want to be "that guy" - but is development continuing on this project? I knew this was a work in progress, and there isn't much that needs to be fixed, but the things that are still hanging are preventing me from enjoying my keyboard fully.

obra, 2023-03-04:
The Numlock key is a hardcoded "special feature" that both predates and
overrides custom color schemes - We're working on changing that, but don't
have a date for it yet.

closed 2024-02-26 by obra