keyboardio/Chrysalis #1243: Support setting default LED plugin from Chrysalis / Heatmap support

lldata, 2022-12-23:
**Is your feature request related to a problem? Please describe.**
I think the LED button is toy, and for daily productive use of the keyboard I just want a single LED layout as my default
I prefer Heatmap, which isn't even included in the default firmware btw

**Describe the solution you'd like**
Add a feature in Chrysalis, where it possible to chose the default LED mode a list of plugins
Add heatmap to default firmware

**Describe alternatives you've considered**
I can flash my own firmware to get this, but I think it would be great if it was more accessable

obra, 2022-12-23:
It *is* possible to set the default LED layout from the chrysalis
preferences screen, although you need to count how many taps of "LED" you
need to do to figure out the right number for the led mode to set on the
preferences screen.

best,
jesse

lldata, 2023-04-26:
Ok, but can heatmap get included in the default firmware?