keyboardio/Chrysalis #1252: Presets for common keymaps

tremby, 2023-02-07:
**Is your feature request related to a problem? Please describe.**
I bought the Model 100 and some Dvorak keycaps. I had not used Chrysalis before, since when I got my M01s it wasn't yet ready, and by the time it was, I had customized my firmware quite extensively.

In setting up my Model 100 I figured I'd try Chrysalis. I thought that since there's first-class support for Dvorak (I bought a keycap set after all) there'd be an easy preset for Dvorak.

I thought initially that this would be the "keyboard layout" option in the preferences pane but it looks like this does something else.

I searched around Chrysalis for quite some time for a "load preset" feature and could not find it. I did web searches too, thinking it must exist, but couldn't find anything.

**Describe the solution you'd like**
Assuming the feature doesn't already exist: an option to load a preset layer or set of layers, which would have some provided options like Qwerty and Dvorak and Colemak. Maybe it could accept custom presets too.

On the other hand if this feature *does* already exist, I think it needs to be somewhere more obvious, because I can't find it.

**Describe alternatives you've considered**
Well I set each key one by one, which was awkward. I opened tickets with my perceived bugs and ideas for improvement related to this separately.

obra, 2023-02-07:
https://shop.keyboard.io/pages/changing-the-model-100s-layout-to-dvorak-colemak-dh-or-something-else
explains the process. It's linked from
https://shop.keyboard.io/pages/model100

tremby, 2023-02-07:
OK, "backup and restore" is not a good name for that button. I saw it, but I don't want to back up yet, and have nothing to restore, so I did not consider clicking it.

I suggest changing it to two buttons. "Load layout" would do what the current button does, and "save layout" or "backup layout" would go straight to the export option.

I might also suggest linking that article from the Chrysalis readme, and also from the [Keyboardio FAQ page](https://shop.keyboard.io/pages/faqs) (which incidentally right now is only about the Atreus other than one out-of-date mention that the M100 is not yet in production!)