keyboardio/Chrysalis #1253: "Layer shift when held" option isn't selectable

tremby, 2023-02-07:
**Describe the bug**
I see an option "Layer shift when held" under the "Layers and keymaps" accordion but I cannot select it.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to layout editor
2. Click on a key
3. Open the "layers and keymaps" accordion
4. Click the "type" dropdown, see unselectable option "layer shift when held" option.

**Expected behavior**
It either is selectable and does something, or there's some clue on screen as to what I need to do to enable it, or why it's unavailable.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Chrysalis Version: 0.12.0

**Additional context**
Model 100.

I was looking for a way to trigger a layer while holding both function keys. I wondered if this might help but couldn't figure out how to enable it. I'll open another ticket asking whether there is a way to do what I want.

fluffy-critter, 2023-02-12:
From other threads it seems that this is functionality enabled by a plugin of some sort; what plugin this is and what it does (and how it differs from “shift to layer”) absolutely should be documented.

TrueFalcon, 2023-02-12:
"Layer Shift" and "Shift To Layer" mean the same thing; you always have to hold the key to stay on the layer and releasing it puts you back where you came from. "Layer shift when held" can't ever be selected but it can be lit up! I'll show you how!

"Layers and keymaps" set the primary action on a key and it becomes the only thing the key will do. You can't set an "Secondary action" on such a key. If the key already has a letter on it, say 'L', then selecting "Shift To Layer" under Layers and keymaps will replace the letter; it won't do 'L' anymore.

Now "Seconday action". This can make a key do 2 things: Tap for the primary action and hold for the secondary.

Under "Secondary action" select "Layer shift" to "Three" or '3' if you haven't renamed layers as I have. After this selection, look under "Layers and keymaps" it says "Layer shift when held." Click it and observe that "Layer shift when held" is lit up but you still can't select it in the drop down. It's a warning that if you now select any thing here, you will wipe out the 'L' and the secondary action.

[image]

Perhaps this could have been explained somewhere in the Chrysalis User Manual ...