keyboardio/Chrysalis #1256: Chrysalis Does Not Save Layer #0

gavinhoward, 2023-02-14:
**Describe the bug**
I used Chrysalis to create [this `Layout.json`][1]. As you can see, Layer 0 is a modified Dvorak.

Nevertheless, after saving the configuration, the keyboard still acts like it has Qwerty as Layer 0.

**To Reproduce**
Steps to reproduce the behavior:
1. Download the latest Chrysalis release AppImage.
2. Make it executable.
3. Run it.
4. Import [this `Layout.json`][1].
5. Save the configuration.
6. Check for Qwerty

**Expected behavior**

I expected Dvorak.

**Screenshots**

I wasn't able to save a bundle. Chrysalis froze when I tried.

**Desktop (please complete the following information):**
 - OS: Gentoo Linux, up-to-date
 - Chrysalis Version: 0.12.0

**Additional context**

I flashed the latest firmware before saving the configuration.

The other layers seem to work fine.

[1]: https://gavinhoward.com/uploads/Layout.json

obra, 2023-02-14:
If you exit Chrysalis and restart, what does it show for Layer 0? If you go
into the keyboard settings, what do you see for the 'default layer' setting?

gavinhoward, 2023-02-14:
It shows Dvorak, and the default layer is Layer 0.

Another weird thing is that Layer 1, which is supposed to be my Qwerty layer, switched to showing Dvorak.

obra, 2023-02-14:
Do you have a second computer you can plug the keyboard into? I'd really
like to get a debugging bundle and also to see if the QWERTY behavior is
replicated on other devices.

gavinhoward, 2023-02-14:
I do!

It is an Ubuntu 20.04 laptop.

The bundle is [chrysalis-debug.zip](https://github.com/keyboardio/Chrysalis/files/10737693/chrysalis-debug.zip).

When I compared the two computers, something interesting happened. The six images below are the keymaps on the Gentoo machine.

[image]
[image]
[image]
[image]
[image]
[image]

(Adding an extra paragraph for space between screenshot sets...)

The six below were taken on the Ubuntu machine.

[image]
[image]
[image]
[image]
[image]
[image]

I don't know why they disagree, but the one on the Ubuntu machine matches how the keyboard actually acts, I think.

I don't know what I'm doing wrong, but maybe I can import my JSON layout on the Ubuntu machine and make it work that way until the problem is debugged.

gavinhoward, 2023-02-14:
By the way, both sets of screenshots were taken right after plugging in the keyboard and opening Chrysalis.

obra, 2023-02-14:
Is the Chrysalis build you're using on Gentoo downloaded from our Github or
from their build?

obra, 2023-02-14:
Oh. In the settings panel in Chrysalis, what have you set the host keymap
to on each OS?

gavinhoward, 2023-02-15:
The build on both is downloaded from GitHub.

The keymap was different. I was stupid and didn't realize that that was supposed to be the keymap on the computer. PEBKAC as usual. I'm going to redo my layouts with it set the way it's set on the Ubuntu machine. If it works, I'll close this.

gavinhoward, 2023-02-15:
As I thought, it was PEBKAC. I apologize.

closed 2023-02-15 by gavinhoward