keyboardio/Chrysalis #1257: Atreus not found on Fedora 36

TrevorFSmith, 2023-02-23:
**Describe the bug**
Chrysalis does not find a new Keyboardio Atreus

**To Reproduce**
Steps to reproduce the behavior:
1. Plug in the Atreus to a Fedora Linux laptop
2. Run the 0.12.0 AppImage
3. See the "No keyboards found" message

**Expected behavior**
I expected the keyboard to be found.

After reading a few closed issues here I tried a few things:
- Added my user account to the dialout group, then rebooting
- chmod 666 /dev/ttyACM0 (which appears and disappears when I plug and unplug the keyboard)
- adding [these udev rules](https://raw.githubusercontent.com/keyboardio/Kaleidoscope/master/etc/60-kaleidoscope.rules)
- ran Chrysalis as root

No luck so far.

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Fedora 36 running i3wm on a Razor Blade 15
 - Chrysalis Version: 0.12.0

**Additional context**
Add any other context about the problem here.
[chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/10810153/chrysalis-debug.json.txt)

obra, 2023-02-23:
Weirdly, /dev/ttyACM0 has an intel vendor id associated with it. Is the
keyboard plugged into a docking station or directly into the computer?

After you added the udev rules, did you `udevadm control --reload-rules &&
udevadm trigger`?

I don't have a current fedora machine handy to check, but do you have
devices under /dev/serial/?

TrevorFSmith, 2023-02-23:
The keyboard's cable is connected directly to a USB-A port on the side of the laptop.

I did run a udev reload rules before but just in case I flubbed it I ran the command you wrote to no avail. I rebooted just in case with no change.

When I plug in the keyboard I see a link appear:
/dev/serial/by-path/pci-0000:00:14.0-usb-0:1:1.0 -> ../../ttyACM0

Just to check that it's this laptop/OS and not the keyboard, I ran Chrysalis on a similar Razor Blade with Windows 11 on it and it was found with no problem. I successfully updated the firmware and loaded the Colemak layout, too, so I am not blocked on using the keyboard, I just can't modify it from my main Fedora laptop.

Just in case it's useful, I exported another Chrysalis debug file on my Fedora laptop:
[chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/10818479/chrysalis-debug.json.txt)

obra, 2023-02-25:
I installed a Fedora 36 box and managed to reproduce the problem. This
is...super weird.

TrevorFSmith, 2023-02-25:
Out of curiosity I connected the Atreus to an AMD tower machine running Fedora to see if it has the same issue. Chrysalis still reports that no keyboard is found.

In the serial port list /dev/ttyACM0 has different product and vendor IDs than listed on the laptop.

[chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/10830127/chrysalis-debug.json.txt)

obra, 2023-02-25:
What I'm seeing on fedora can be reproduced with this commandline:

npx @serialport/list -f jsonline

On fedora, it's showing the vendorid and deviceid for your usb controller.
On macos, it's correctly showing the device's vendorid and deviceid.

I'd love to know if this is broken on fedora 35 or fedora 37

TrevorFSmith, 2023-02-25:
Ok! I connected the Atreus to an Intel NUC running Fedora 37 and Chrysalis found it with no problem. It's not a one-to-one comparison with earlier info since it's a different machine than the Razor Blade and my AMD tower but it's still possible that the situation globally changed with the new version of Fedora.
[print-control-1.chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/10831923/print-control-1.chrysalis-debug.json.txt)

obra, 2023-02-27:
Confirmed that the issue is specific to node-serialport and Fedora 36. Bug
filed at https://github.com/serialport/node-serialport/issues/2604

TrevorFSmith, 2023-02-27:
That's wild. I'm glad that it's no longer a mystery and that it's fixed by an OS update.

closed 2023-02-27 by TrevorFSmith

obra, 2023-07-02:
this ended up turning out to be caused by a regression in systemd that showed up in fedora 36 and debian 11 and has since been patched

TrevorFSmith, 2023-07-02:
Sweet! Thank you for looking into it.