keyboardio/Chrysalis #1269: Error when Crysalis is trying to update

Random-Man81, 2023-05-12:
[chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/11467988/chrysalis-debug.json.txt)
[image]
Error about checksum mismatch when trying to update to 0.13.2

thunderkeys, 2023-05-21:
I get the same error (same hashes as well).

necsoiu, 2023-05-21:
I have the same problem.

TrueFalcon, 2023-05-22:
I gave up on 13.# and went back to 12.0. It works fine. Awaiting next higher version...

obra, 2023-07-31:
I've finally tracked this one down to a race condition caused by the "master" build overwriting the release build of the exact same version at publication time. I've manually repaired the build information, but we're now running into an Authenticode signature issue. Coupled with the fact that our old certificate expired yesterday and new certificates are only being issued in a way that makes github-based builds...more complicated and expensive, I'm not yet sure how we're going to sort this out.

obra, 2024-02-26:
Chrysalis now runs in-browser using WebSerial and WebUSB and this functionality has been rewritten, so I'm closing out this issue as obsolete. Please don't hesitate to open a new issue if https://chrysalis.keyboard.io exhibits the same behavior

closed 2024-02-26 by obra