keyboardio/Chrysalis #1272: Key R won't work (can't get char R by pressing R key)

artkpv, 2023-05-26:
**Describe the bug**

R key won't work.

**To Reproduce**
Steps to reproduce the behavior:

0. I open any text editor or console.
1. I press at Atreous keyboard R key.

Result: I don't see any character appearing, not R, not other.

Also, when I move connection of the cable to my keyboard the 'r' char appears many times (indefinetly) till I press some other key.

**Expected behavior**
Expected: I see 'r' character at the screen

**Desktop (please complete the following information):**
 - OS: Archlinux
 - Chrysalis Version: Not sure. 0.12.0

**Additional context**
[chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/11574051/chrysalis-debug.json.txt)

TrueFalcon, 2023-05-26:
Try switching the R keyswitch with some other letter and see if the problem moves with the switch. If yes, replace the switch. Also move the cable and see if the swap changes anything.
Press the hamburger at top left and the Chrysalis version should be displayed below the exit button.

artkpv, 2023-05-27:
@TrueFalcon  Thank you for your answer. I tried cleaning the connection hole at the keyboard and used my older cable that came with the keyboard. The R key now works. Perhaps it was some dirt inside. I will close this issue then.

Another thing. Perhaps there is quick answer. I tried to flash new firmware and see this error (below). But I _gave_ those permissions. Steps: 1. plugin the keyboard. 2. Run `sudo chmod 666 /dev/ttyACM0 ` . 3. Open Chrysalis. 4. Open the page to update. 5. Click Update. 6 Press ESC. 7. Click to run the update. Result: It goes up to 3 point there and then gets to this error message.

Step 4:
[image]

Step 5:
[image]

Result:
[image]

closed 2023-05-27 by artkpv