keyboardio/Chrysalis #1276: Automatic Update Fails: Error during update: Error: sha512 checksum mismatch

Zian, 2023-06-03:
**Describe the bug**
Chrysalis cannot update itself. It says the file it got is bad.

**To Reproduce**
Steps to reproduce the behavior:
1. Start Chrysalis.
2. Wait for the new version to download by watching the progress bar in the lower left corner.

**Expected behavior**
Software update installs.

**Actual Result**
A red floating window appears that says:

Error during update: Error: sha512 checksum mismatch, expected 8eqU1ytk1LOlQcsjaaOeLom3ci2VEgdHRfqdYi3nXFj7UDOsSdvgaVqbNddqXoPQfSQHDWnN3C5fAvK07XhmTg==, got Skw28NoMtmr5rEftlvTFYNoYSffphHlGnv/DNLy1P7wmxym/s8V43FeWHRs20owsyxoJpuLdhv9RZQlY1lWj/w==

**Screenshots**
[image]
[chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/11643523/chrysalis-debug.json.txt)

**Desktop (please complete the following information):**
 - OS: Microsoft Windows Version 22H2 (OS Build 19045.2965)
 - Chrysalis Version: 0.12.0

**Additional context**
I asked GitHub for the file the same way that Chrysalis does by re-sending the HTTP requests that Fiddler intercepted. Then, I compared its SHA512 hash against the SHA512 hash of the file that GitHub provided when I downloaded the file using a web browser. The two files have the same hash.

However, neither file hash matches the 2 hashes displayed in the error message.

My files' SHA-512 hashes were both "4A4C36F0DA0CB66AF9AC47ED96F4C560DA1849F7E98479469EFFC334BCB53FBC26C729BFB3C578DC57961D1B363A8C2CCB1A09A6E2DD86FF51650958D485A3FF".

Zian, 2023-06-10:
Probably a duplicate of https://github.com/keyboardio/Chrysalis/issues/1269.

obra, 2023-07-31:
I've finally tracked this down to a race condition caused by a build on master accidentally overwriting the release build. I've manually repaired the release and am in the process of fixing the build system.

closed 2023-07-31 by obra

obra, 2023-07-31:
(And it's currently flagging a code signing issue on the build now that it downloads. digging into that)