keyboardio/Chrysalis #1278: Cannot update firmware

khers, 2023-06-11:
**Describe the bug**
I am attempting to update the firmware on my model 100 and the first pass fails, leaving the keyboard in bootloader mode (Prog key lit red). Chrysalis asks if I want to continue without saving my settings and when I say yes the Chrysalis window goes completely white and nothing else happens. I am starting with a custom firmware where I compiled out space cadet shift (I couldn't get Chrysalis to disable it so removed the code that did it from the firmware).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to update firmware in chrysalis
2. Click on Continue while holding prog
3. Wait for failure
4. Proceed even if settings will be lost.
5.  Chrysalis hangs, leaving the keyboard in bootloader mode

**Expected behavior**
The firmware update to complete, even if settings were lost.

Attached debug bundle is from a failed update before the second attempt. On the second attempt Chrysalis has hung and I cannot create a bundle.
[chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/11715539/chrysalis-debug.json.txt)

**Desktop (please complete the following information):**
 - OS: Pop_OS! 22.04
 - Chrysalis Version: 0.13.2

khers, 2023-06-11:
Initial flashing behavior is exactly the same at #1275

obra, 2023-06-27:
Hi Eric,

I'm sorry for the delay - I was traveling for the last several weeks.
Digging into this, it looks like everything is fine with the serial
communications side of things, but that we start running into trouble when
trying to do the DFU firmware upload.

It looks like you have...a lot going on on your USB bus.  Just as a
debugging step, do you run into the same issue when trying to flash if you
plug the keyboard into a USB port physically on the computer (likely next
to the Focusrite Scarlett)

Does it look like Chrysalis installed the udev rules for Kaleidoscope
keyboards?
Do you have a file called '/etc/udev/rules.d/60-kaleidoscope.rules'? If so,
does it have lines about the Model 100 in it?

khers, 2023-06-30:
The Keyboard has always been plugged directly in the motherboard, I moved it right next to the audio interface and the problem continues.

I do not have the kaleidoscope rules file.  And adding the one from the kaleidoscope repo solved my issue.

obra, 2023-06-30:
I'm glad you got it working, but we *should* be prompting to install those rules at startup if they're not detected. I'm going to reopen this issue to track that problem.