keyboardio/Chrysalis #1279: No keyboards found.

faxm0dem, 2023-06-13:
**Describe the bug**
When running chrysalis, it says "No keyboards found.".
I hooked up both my Atreus and Model 01.

**Desktop (please complete the following information):**
 - OS: Debian 11
 - Chrysalis Version: 0.13.2

**Additional context**
I see the following in the console when chrysalis starts up:
```
[638195:0613/090819.070304:ERROR:bus.cc(399)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
[638195:0613/090819.070749:ERROR:bus.cc(399)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
```

faxm0dem, 2023-06-13:
Both `/dev/ttyACM` devices show up on Linux by the way

faxm0dem, 2023-06-13:
I think something must have messed up the firmware in a previous chrysalis version. Both my model 01 and atreus fail to be recognised, but my model 100 is.

obra, 2023-06-21:
> My Model 100 *is* recognised by chrysalis 0.13.2, but invites me to
> upgrade its firmware.
> I am wary to do so, given the current state of my 2 other keyboards.

So sorry for the delay in our reply. If you back down to an earlier
chrysalis version, does it see your devices?

What are the permissions you see on the /dev/ttyACM devices? How about the
/dev/serial/by-* devices?

faxm0dem, 2023-06-22:
I tried that to no avail, still "no keyboards found"
About the permissions, the are `rw` for group `dialout` which I am a member of.

obra, 2023-06-23:
What do you see in `find -type f /dev/serial`?

Are the devices plugged directly into the computer or through a hub or
switch?

faxm0dem, 2023-06-26:
I tried both.
My Model 100 pops up in chrysalis even though I'm using a thunderbolt dock.
My Atreus is plugged directly using a regular USB port on my laptop.
Both show up as following in `/dev/serial/`:
```
/dev/serial
/dev/serial/by-path
/dev/serial/by-path/pci-0000:00:14.0-usb-0:2:1.0      # <-- Atreus
/dev/serial/by-path/pci-0000:3b:00.0-usb-0:1.2.2:1.0  # <-- Model 100
```

obra, 2023-06-28:
I'm trying to repro this but haven't managed so far. Can you grab a "report an issue" bundle from the sidebar menu in Chrysalis after plugging both a good and a bad keyboard in?

And just to be sure, this is the 0.13.2 appimage downloaded from our github repo?

obra, 2023-06-28:
I've managed to repro this and tracked it down to https://github.com/serialport/node-serialport/issues/2604, which upstream looks like they're treating as "wontfix" :/

I'm not yet sure if we've got a sane workaround.

obra, 2023-06-28:
Got it. It's a bug in Debian, caused by a bug in systemd. https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1035094

obra, 2023-06-28:
(I verified that manually applying the fixes to the udev rules described therein gets the Atreus to start showing up fine for me on a fresh Bullseye 11.7 install.

faxm0dem, 2023-06-28:
Awesome, thanks for the digging !
I vaguely remember having had to fiddle with some udev and/or systemd config to make my FTDI adapter show up correctly in `/dev/by-id/...`... maybe that's related

faxm0dem, 2023-06-28:
I tried applying [this](https://github.com/systemd/systemd-stable/commit/451ba55fecd8b494add2001b3ca3c1915c8fd655) patch and reloading `udev` rules to no avail.
I guess I'll wait for the debian fix - thanks @obra !

faxm0dem, 2023-06-28:
I just had a thought : why is my 100 still recognised ?

obra, 2023-06-28:
We use a different technique for the Model 100 because its firmware update
method doesn't use serial. My recollection is that we can't use that method
to find the M01 and Atreus serial ports, but I'm not 100% on that.

obra, 2023-06-28:
FWIW, the two patches I applied (by hand-typing the code):
*both*
https://github.com/systemd/systemd-stable/commit/451ba55fecd8b494add2001b3ca3c1915c8fd655
and
https://github.com/systemd/systemd/pull/25246

closed 2024-02-26 by obra