keyboardio/Chrysalis #1297: Firmware update freezes with error ENOENT: no such file ... static/build-info.yml

parke, 2023-07-29:
**Describe the bug**

My Linux-based runtime environment does not support mounting FUSE filesystems.  Consequently, I extract the AppImage to an AppDir.

When I try to run the Firmware Update with Chrysalis 0.13.2, I get the following error message:

```
Error: ENOENT: no such file or directory, open '/opt/atreus/atr-0.13.2/usr/lib/chrysalis/static/build-info.yml'
    at Object.openSync (node:fs:600:3)
    at Object.func [as openSync] (node:electron/js2c/asar_bundle:2:1822)
    at Object.readFileSync (node:fs:468:35)
    at t.readFileSync (node:electron/js2c/asar_bundle:2:9165)
    at IpcMainImpl.<anonymous> (/opt/atreus/atr-0.13.2/usr/lib/chrysalis/resources/app.asar/.webpack/main/index.js:8:922535)
    at IpcMainImpl.emit (node:events:513:28)
    at EventEmitter.<anonymous> (node:electron/js2c/browser_init:2:82058)
    at EventEmitter.emit (node:events:513:28) {
  errno: -2,
  syscall: 'open',
  code: 'ENOENT',
  path: '/opt/atreus/atr-0.13.2/usr/lib/chrysalis/static/build-info.yml'
}
```

Additionally, Chrysalis freezes until I exit Chrysalis by closing the window.

It appears the missing file may be present, but in a different directory:  
`/opt/atreus/atr-0.13.2/usr/lib/chrysalis/resources/static/build-info.yml`

This error does not happen with Chrysalis 0.12.0.

**To Reproduce**
Steps to reproduce the behavior:
1. Extract the AppImage to an AppDir with `./Chrysalis-0.13.2-x64.AppImage --appimage-extract`
2. Run the AppDir with `./AppRun`
3. Open the hamburger menu.
4. Click on 'Firmware Update'

**Expected behavior**
A firmware update screen should appear.  (As happens with Chrysalis 0.12.0.)

**Desktop**
 - OS: Linux
 - Chrysalis Version: 0.13.2

**Additional context**
Probably not relevant:  My keyboard is an Atreus with firmware 0.10.4+5.

obra, 2023-07-29:
We don’t really support running the production build unpackaged. If you’re
not able to use the appimage as built, you may have better luck building
and running Chrysalis from source.

parke, 2023-07-29:
The extracted AppDir has always worked for me before.  However, I understand not wanting to support Chrysalis as an AppDir.  Please feel free to close this bug.

(Aside: Also, I may continue to indefinitely postpone upgrading the firmware, as I prefer having 10 layers on my Atreus.)

obra, 2023-07-30:
We upgraded to a maintained Electron packaging system (electron-forge), so
a lot of the build infrastructure changed. I'll happily take patches if you
track down the issue, but yeah, we just don't have the resources to support
the AppDir.

parke, 2023-07-30:
If I identify the source of the problem, I will let you know.

closed 2023-07-30 by parke