keyboardio/Chrysalis #1299: On startup, Chrysalis fails to properly detect and unstick a sticky Shift key.

parke, 2023-07-30:
**Describe the bug**
On startup, Chrysalis 0.13.2 fails to properly detect and unstick a sticky Shift key.

**To Reproduce**
Steps to reproduce the behavior:
1.  Start Chrysalis.
2.  Set a key to be the left Shift key.
3. In Modifiers, toggle Sticky on.
4. Save the layout to your keyboard.
5. Exit Chrysalis.
6. Restart Chrysalis.
7. Select the key.
8. Open Modifiers.
9. Note that Chrysalis displays the Sticky toggle as being off.  (It should be on.)
10. In Modifiers, toggle Sticky on.
11. Note that the key has changed from "OSM / LSft" to "# 98083".  (I believe that is the number.  It is blurry on my 720p display.)

**Expected behavior**
When the key is selected (numbers 7 through 9 above), Chrysalis should detect and display the Sticky toggle in the 'On' position.

**Desktop**
 - OS: Linux
 - Chrysalis Version: 0.13.2

**Additional context**
Keyboard is an Atreus with the 10.4 firmware.

edysli, 2023-10-31:
I have the same issue with a Model 100 (firmware 0.91.1+74), Chrysalis 0.13.2 and Linux (kernel 6.2.0). The key code for "OSM/LSft" shows 49154 though (in the "custom key code" section on the right panel). The displayed key on the keymap is correct "OSM/LSft".

Having the "OSM/LSft" key selected, there is also a weird behaviour if I click any of the toggles under "Modifiers": the key displays the keycode instead, and this code changes every time I click one of the toggle switches. The toggled switch also never flips to its "enabled" state...

gsuzor91, 2024-03-14:
The same issue as the original post exists for me in the web-based version at https://chrysalis.keyboard.io/ with an Atreus. It was broken a few days ago and remains broken with the new version released today - 2024.0314.1919

obra, 2024-03-18:
I think I've found and fixed the issue. The change should be live on
chrysalis.keyboard.io

commit 24621546 (2024-03-18): Refactor SpecialModifiers component to finish fixing an issue where OSM shift keys weren't editable after save

- Updated `isInCategory` checks to use `key.code` for consistency in `SpecialModifiers.js`
- Simplified `isModifier` check and used it to streamline the disabled condition for the OneShot switch

Fixes #1299
closed 2024-03-18 by obra