keyboardio/Chrysalis #1304: Firmware update failed

EmeraJade, 2023-08-27:
**Describe the bug**
I tried to flash the newest firmware update and got a warning message that it'll have to reset to factory settings. I quickly saved my settings, then continued; which then gave me this error:
> Problem encountered while flashing
> Unable to find the kevboard. Please file an issue.
Now the keyboard cannot be found anymore and does not work at all 😭 Chrysalis itself also cannot exit that warning message, unless I quit it fully.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Firmware update
2. Followed Instructions
3. Error appears

**Expected behavior**
Usually the firmware updates went over smoothly and without errors. I expected the same to happen as always.

**Screenshots**
If applicable, add screenshots to help explain your problem. If possible, please also include the debug bundle you can create on the "Report a problem" screen. If, for some reason, you are unable to access that screen, please press `Control + Shift + I` (Windows or Linux) or `Command + Option + I` (macOS) to bring up the developer console. Right-clicking inside it will allow you to save its contents to a file - please attach that to the bug report.
[image]
[image]

**Desktop (please complete the following information):**
 - OS: macOS Ventura 13.4.1 (c) 
 - Chrysalis Version: Version 0.13.2 (0.13.2)

**Additional context**
It's been a while since I updated the firmware, if that matters. Please help :((

EmeraJade, 2023-08-27:
Okay, I managed to get the keyboard working again! I had to flash it via chrome. But still, even after a factory reset, the firmware update will _not_ work. 
I also had to completely recreate my layout settings, _and_ every time i try to export them somehow (both as json or just as layout card PDF) _it will fuck it up???_
I just quit Chrysalis for now and won't touch it anymore, but something is very wrong still.

obra, 2023-08-28:
Notes from our debugging session yesterday:

It appears that flashing to a recent (0.9.1 or 0.9.2) firmware build,
whether by Chrysalis or by the WebDFU updater resulted in a soft-bricked
keyboard. The firmware itself wouldn't boot, although the bootloader
remained functional.

@EmeraJade is going to be out of town for a bit, but we're going to try
some further debugging techniques when they're back.

merlindru, 2023-09-02:
[image]

I'm getting a similar issue, I think, but in some cases chrysalis just goes completely white after the "Reconnecting" step. My keyboard is soft-bricked as well now. When I hold down the prog key, it goes red before flashing -- before, it was green

rafftre, 2023-09-09:
> EmeraJade

I've the same problem under macOS Ventura 13.5 with Chrysalis 0.12.0. My Atreus keyboard seems soft-bricked now.

merlindru, 2023-09-09:
What if you try flashing it through [chrysalis.keyboard.io](https://chrysalis.keyboard.io/)? @rafftre

merlindru, 2023-09-09:
And I was advised to turn on debug logs and post them in a new issue

rafftre, 2023-09-09:
I didn't understand why, but under macOS I could no longer connect the keyboard. I couldn't even see the serial port anymore (for trying with Kaleidoscope or Arduino IDE). Later I switched to a Windows machine where the keyboard was detected successfully from both the system and Chrysalis. I haven't tried updating the firmware again though, so I'm keeping it that way for now 😃 
After this the keyboard is working as before under macOS.

I apologize if this I introduced some overhead on this issue

obra, 2023-12-21:
It's taken a while, but I *believe* we've finally figured out what was going wrong. The recent Chrysalis 0.13.3 and Kaleidoscope 0.92 firmware versions -should- fix it. If they don't, I'd love to know.

closed 2023-12-21 by obra