keyboardio/Chrysalis #1305: "Communication timeout" with Chrysalis(0.13.2) after recent firmware upgrade

CodingFutu, 2023-08-31:
Hi,

Just wanted to report. I am getting "Communication timeout" when trying to connect keyboardio model 100 to Chrysalis(0.13.2). This started to happen a few days ago after the firmware upgrade.  The keyboard itself is working fine. thanks

[image]
[image]

kjon-life, 2023-09-23:
same here--trying to get my model 01 working 
[image]

obra, 2023-09-25:
I'd love to know if https://chrysalis.keyboard.io in a chrome-based browser
works better for you.

(Also, after you run into this issue, please go into the sidebar menu and
generate a debugging bundle. Without that, we don't have much to go on.)

connanp, 2023-11-21:
[chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/13431747/chrysalis-debug.json.txt)

I'm also experiencing the error. I had changed my layout with colors, used the copy and paste layer functions from #0 to #3. After saving, the communication timeout error began to appear and connecting to the device no longer works. The web version is stalled on "reading from device".

The changes I made saved to the keyboard.

palthedog, 2023-12-10:
Same here.
https://chrysalis.keyboard.io/ doesn't work for me as well.

obra, 2023-12-11:
I'd love to know if the 0.92 firmware build works better for folks running into this issue

palthedog, 2023-12-11:
For me, 0.92 seems to be the root cause of the issue.
I updated the firmware version to 0.92 yesterday. After that, my Chrysalis could never connect to my Model 100 correctly.
I've tried older Chrysalis (0.13.0) but it didn't resolve the issue.

palthedog, 2023-12-11:
As far as I checked the log, my keyboard doesn't respond to a request something like
```
{
    "id": 7,
    "command": "keymap.custom",
    "args": []
}
```

obra, 2023-12-11:
can you attach a full debug bundle?

palthedog, 2023-12-11:
Here you are.
[chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/13634487/chrysalis-debug.json.txt)

obra, 2023-12-12:
@palthedog If I say "serial terminal" is that something that means something to you?

The Arduino IDE has a built in serial terminal and I'd *really* like to see what your keyboard says when you send the commands 

*help*
*version*
*keymap.custom*

palthedog, 2023-12-12:
I'm not familiar with Arduino but here is the result.
I hope it helps you to resolve the issue.

```
macros.map
macros.trigger
mousekeys.scroll_interval
mousekeys.init_speed
mousekeys.base_speed
mousekeys.accel_duration
mousekeys.warp_grid_size
palette
colormap.map
idleleds.time_limit

.
0.92.0+77 
.
65535 30 31 32 33 34 76 231 44 70 35 36 37 38 39 49 53 20 26 8 21 23 43 224 228 40 28 24 12 18 19 46 75 4 22 7 9 10 41 225 229 42 11 13 14 15 51 52 78 29 27 6 25 5 17451 226 226 17451 17 16 54 55 56 45 65535 58 59 60 61 62 65535 65535 40 18614 63 64 65 66 67 68 43 65535 65535 2102 2103 0 65535 65535 17497 18637 18613 2095 2096 47 48 69 74 65535 65535 2086 2087 2097 101 65535 17496 76 80 81 82 79 0 0 77 70 73 65535 65535 49 65535 65535 17492 65535 18658 18666 18665 0 49 2097 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 24576 65535 95 96 97 86 17492 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 92 93 94 87 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 89 90 91 46 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 98 55 85 84 40 65535 30 31 32 33 34 76 44 44 70 35 36 37 38 39 49 53 20 26 8 21 23 43 224 228 40 28 24 12 18 19 46 75 4 22 7 9 10 41 225 229 42 11 13 14 15 51 52 78 29 27 6 25 5 17451 226 226 17451 17 16 54 55 56 45 65535 30 31 32 33 34 76 44 33 70 35 36 37 38 39 49 53 20 26 8 21 23 43 224 32 40 28 24 12 18 19 46 75 4 22 7 9 10 41 225 31 42 11 13 14 15 34 37 78 29 27 6 25 5 17451 226 30 17451 17 16 35 36 34 37 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 
.
65535 30 31 32 33 34 76 231 44 70 35 36 37 38 39 49 53 20 26 8 21 23 43 224 228 40 28 24 12 18 19 46 75 4 22 7 9 10 41 225 229 42 11 13 14 15 51 52 78 29 27 6 25 5 17451 226 226 17451 17 16 54 55 56 45 65535 58 59 60 61 62 65535 65535 40 18614 63 64 65 66 67 68 43 65535 65535 2102 2103 0 65535 65535 17497 18637 18613 2095 2096 47 48 69 74 65535 65535 2086 2087 2097 101 65535 17496 76 80 81 82 79 0 0 77 70 73 65535 65535 49 65535 65535 17492 65535 18658 18666 18665 0 49 2097 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 24576 65535 95 96 97 86 17492 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 92 93 94 87 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 89 90 91 46 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 98 55 85 84 40 65535 30 31 32 33 34 76 44 44 70 35 36 37 38 39 49 53 20 26 8 21 23 43 224 228 40 28 24 12 18 19 46 75 4 22 7 9 10 41 225 229 42 11 13 14 15 51 52 78 29 27 6 25 5 17451 226 226 17451 17 16 54 55 56 45 65535 30 31 32 33 34 76 44 33 70 35 36 37 38 39 49 53 20 26 8 21 23 43 224 32 40 28 24 12 18 19 46 75 4 22 7 9 10 41 225 31 42 11 13 14 15 34 37 78 29 27 6 25 5 17451 226 30 17451 17 16 35 36 34 37 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 65535 6

```

Let me note that a single `keymap.custom` command doesn't respond anything even I waited the response for 1 minute.
However, if I send the same message again, I got two responses at once.
So what I actually send was:
```
help
version
keymap.custom
<wait for a while>
keymap.custom
```
and also I can reproduce this behaviour again and again like:
```
<send keymap.custom>
<nothing happen>
<send keymap.custom>
.
65535 30 31 32 33 ......
.
65535 30 31 32 33 ......
<send keymap.custom>
<nothing happen>
<send keymap.custom>
.
65535 30 31 32 33 ......
.
65535 30 31 32 33 ......
```

obra, 2023-12-12:
That's spectacularly useful. Thanks!

tlyu, 2023-12-12:
As I mentioned in Discord, but will put here for the record: the first line, including trailing space, plus the 5 characters CR LF period CR LF, is a multiple of 64. The truncated second line, assuming there's no line termination, is also a multiple of 64. This confirms a suspicion that I had about us needing to send ZLPs in some cases to make sure the host sees the output if it's a multiple of 64 bytes.

tlyu, 2023-12-12:
(The total length of those two replies is 4096, a nice round power of 2, which is also 64 times 64. So now we've reverse engineered the buffer size that Windows and/or the Arduino serial monitor is using there.)

tlyu, 2023-12-13:
Awaiting further confirmation, but it seems likely that the combination of keyboardio/Kaleidoscope-Bundle-Keyboardio#51, keyboardio/ArduinoCore-GD32-Keyboardio#52, and keyboardio/ArduinoCore-GD32-Keyboardio#53 fixes these issues.

tlyu, 2023-12-13:
For anyone affected by this problem who hasn't seen this on Discord, this is a new snapshot that might help https://github.com/keyboardio/Chrysalis-Firmware-Bundle/releases/tag/v0.92.1-snapshot

tlyu, 2023-12-14:
Preliminary feedback implies that these ZLP fixes likely either fix these issues, or substantially mitigate them. Thank you to everyone who's tested so far, and please do try out the snapshot if you haven't already.

closed 2024-02-26 by obra