keyboardio/Chrysalis #1306: Firmware update screen claims to update to 0.92, but actually installs 0.91

cflee, 2023-09-03:
**Describe the bug**
Firmware update claims that updating will go to 0.92 when it ends up installing 0.91.1+74.

Clicking on "What's New in 0.92.0?" results in 0.91.1 changelog.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to a Model 100
2. Click on menu, Firmware Update

**Expected behavior**
Firmware update version being displayed and offered should be 0.91.1+74 instead of 0.92.

**Screenshots**
[image]
[image]

[chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/12505684/chrysalis-debug.json.txt)

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Chrysalis Version: 0.13.2

**Additional context**
I discovered this while updating my keyboard that was previously on 0.90.x firmware (some snapshot version), after doing an update to Chrysalis 0.13.2 by downloading the setup exe (was previously on 0.11.x). The screenshots above are taken after running the update from Chrysalis 0.13.2 that claimed to bring it to 0.92.

obra, 2023-09-03:
We pulled the 0.92 update after a report of adverse behavior from a Model
100 user. I suspect there's a caching issue causing it to report the
removed 0.92 version.

shofutex, 2023-12-20:
I receive the same behavior trying to update to 0.92.1.

obra, 2023-12-20:
If you're up for trying what will be released as Chrysalis 0.13.3 tomorrow,
it should fix this issue:
https://github.com/keyboardio/Chrysalis/releases/tag/v0.13.3

shofutex, 2023-12-20:
Yes, that fixed the issue for me. Thanks!

closed 2024-02-26 by obra