keyboardio/Chrysalis #1310: Globe Key (🌐) Not Working on macOS Sonoma

nlordell, 2023-09-28:
**Describe the bug**
Globe key no longer seems to work on macOS Sonoma.

**To Reproduce**
Steps to reproduce the behavior:
1. Save a layout with the 🌐 key (under "Mac/iPad Keys” section)
2. Connect keyboard to a computer on macOS Sonoma and press the configured 🌐 key, **nothing happens** (I expect the emoji picker to open given my current configuration - the macOS default)

**Expected behavior**
Expect the 🌐 key to open the emoji picker. 

**Screenshots**
[chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/12746924/chrysalis-debug.json.txt)

Keyboard and emoji picture should be configured correctly:
[image]
[image]

I tested other macOS keys (lock screen, exposé, mission control) and they work well.

**Desktop:**
 - OS: macOS 14.0
 - Chrysalis Version: 0.13.2
 - Firmware Version: 0.91.1+74
 - Keyboard Model: 01

**Additional context**
I’m not 100% sure, but I remember the globe key working under macOS Ventura (13.x). I unfortunately don’t have access to a computer running that version of macOS anymore to test if my memory is correct.

roijalbaker, 2023-10-02:
I can partially replicate this bug. I am also already on Sonoma, my globe key gives me a list of keyboards to choose from. The behaviour I see is:

- If I hit the 🌐 key on the laptop, it shows me the list.
- If I hit the 🌐 key on the Keyboardio, it doesn't show the list.
- If I hit the 🌐 on the laptop keyboard and then use the 🌐 key on the Keyboardio, I can use it to cycle through the list.

I tried mapping the ctrl key to globe in the Mac keyboard settings, then hitting ctrl does give me the list.

[image]

roijalbaker, 2023-10-02:
@nlordell I do have a tip, you can map a key to ctrl+cmd+space, that will give you an emoji picker (though it disappears after having picked an emoji)

obra, 2023-10-02:
There are two different ways the globe key gets used - on its own and as
part of a chord.

In a situation where Globe on its own doesn't work, does Globe-F
full-screen the current window? IIRC "tap globe on its own" is not
something that ever worked for external third-party keyboards like ours.

nlordell, 2023-10-04:
Another interesting fact, `🌐 + e` seems to bound by default to opening the emoji picker (🎉), but can’t seem to find a way to configure a single key to send the `🌐 + e` chord through in Chrysalis.

NicholasSterling, 2024-05-21:
The Globe key does nothing when I press it on the Model 01, but works fine when I press it on the laptop keyboard.  No emoji picker, no quick notes or notification center.  If I type  🌐 + q while a terminal window has focus, I get an ordinary q on the command line.  The globe key assignment used to work for me, but it has stopped working, and I'm not sure when.  I tested another mapping on that key to make sure the switch still works.  As of May 2024, is the Globe key assignment working for others?  I'm using the latest keyboard firmware (0.92.6+116) and latest Mac OS (Sonoma 14.5), Apple Silicon.

obra, 2024-05-21:
Digging into this more, what I'm seeing is that the globe key mapped through Chrysalis acts as " 🌐"  when tapped twice to start dication, but *not* as the globe *modifier* or when configured to open the emoji picker- Reading up elsewhere, this is consistent with what other keyboard firmwares are seeing. Apple isn't exposing globe-as-modifier through standard USB peripherals. 

It appears that the universally recommended workaround is to map "Caps lock" to "fn" for a given keyboard in system settings:

[image]

It looks like what's going on with apple's first-party keyboards is that they have a custom USB HID device descriptor that maps 'globe' to HID page 0xFF key 0x03. Without adding a custom apple endpoint, I'm not convinced that any keyboard will be able to generate the true globe key.

If your goal were just the emoji picker, mapping a key to Control + Cmd + Space would do it:
[image]

It's far from ideal, but looks like that's what apple gives us these days.

closed 2024-05-21 by obra

NicholasSterling, 2024-05-21:
Thanks -- mapping Caps Lock to fn for the Model 01 in 'System Settings / Keyboard / Custom modifier keys' and mapping a physical key to Caps Lock in Chrysalis gets me all the Globe key behavior I was looking for:  🌐 + q for quicknotes,  🌐 + n to open the notification center,  🌐 + e (or just  🌐 🌐 )to open the emoji picker, etc.  Perfect.

lldata, 2024-05-21:
> Digging into this more, what I'm seeing is that the globe key mapped through Chrysalis acts as " 🌐" when tapped twice to start dication, but _not_ as the globe _modifier_ or when configured to open [...]

It worked for me until some point in 2023. I have been upgrading both MacOS and Crysalis, so hard to say which changed.
But @obra do you feel sure that this is a change on the MacOS side, since you close the ticket? My fn+space is mapped to globe. I am positive that it used to work, no workarounds required.

obra, 2024-05-21:
I'm pretty sure that things have been changed/tightened up on the macos side.

TurekBot, 2024-10-04:
With <kbd>Control</kbd> + <kbd>Command</kbd> + <kbd>Space</kbd> having [stopped working](https://www.reddit.com/r/firefox/comments/16ykyu1/comment/kpciflb/?utm_source=reddit&utm_medium=web2x&context=3) on Firefox, I've been hurting.

There's been [a Firefox patch to get it fixed](https://phabricator.services.mozilla.com/D195016) for a while now, but it's been almost a year.

Thanks to you good Keyboardio people, today I was able to re-map Caps to Fn and resume emojiing conveniently. 🙏