keyboardio/Chrysalis #1311: Check firmware update yields version [object Object]

Kilomil, 2023-09-30:
**Describe the bug**
Trying to get new chrystalis firmware version for model 100 was very slow and at the end the new version it got from the check was [object Object]

**To Reproduce**
1. I installed the new Chrystalis on top of my old installation.
2. I went to menu Firmware Update
3. wait a while
4. it reports newest version is [object Object]

**Expected behavior**
That I would get a proper version. Or if fetching the version fails, that it would not show me to update to an unspecified version and maybe an error popup.

**Screenshots**
[image]
[image]

[chrysalis-debug.json.txt](https://github.com/keyboardio/Chrysalis/files/12774225/chrysalis-debug.json.txt)

**Desktop (please complete the following information):**
 - OS: win 10
 - Chrysalis Version: [e.g. 0.13.2]

**Additional context**
Add any other context about the problem here.

obra, 2023-09-30:
it does look like you have the most recently released version of the
firmware. (We briefly released in 0.92 and then pulled it after hearing
reports of a possible issue from one customer.)

if this is reproducible, is there anything in the developer console error
log?

Kilomil, 2023-10-01:
Is the snapshot 2 I provided not the developer console error log?
And yes its reproducible. Just tried, same result and the console output is the same as provided in snapshot 2.
I see someone in the discord channel showing a snapshot for them, and I can see they, on their model 100, have firmware version 0.91.0+74, where I only have 0.91.0+68.
https://media.discordapp.net/attachments/492409398493446146/1157642118785925150/image.png?ex=651aab3d&is=651959bd&hm=ea3a71fdc43b7b2e25da6bfa1f37ca02cc99db6d251f4aee823976da4efe8c1d&=&width=858&height=608

Kilomil, 2023-10-01:
I also just tried to uninstall both v12 and 13 of chrystalis, and then re-install v13. Same result.

obra, 2023-10-01:
Indeed it was. So sorry. I'm not sure why it didn't render for me the first
time I looked.

That "file not found" error does kind of hint that something might be
wrong. Can you grab a snapshot of the network panel to see if it can show
off what's not getting loaded?

Kilomil, 2023-10-06:
[image]
I dont see much activity in the network log

Kilomil, 2023-10-06:
[image]
[image]
Managed to capture a few more

obra, 2024-02-26:
Chrysalis now runs in-browser using WebSerial and WebUSB, so I'm closing out this issue as obsolete.

closed 2024-02-26 by obra