keyboardio/Chrysalis #1318: Secondary action modifier alt and control have switched names/places

ahusby, 2024-01-07:
**Describe the bug**
When I assign secondary action control to a key the action of the keypress is alt. When I assign the secondary action alt to a key the action of the keypress is control

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04 LTS with stock Gnome desktop environment and with input source/keyboard layout "Norwegian"
 - Chrysalis Version: 0.13.3 with keyboard layout "Norwegian"

ahusby, 2024-01-11:
[image]

bschwedler, 2024-01-16:
I have also started seeing this, but on a Model M01.

OS: Fedora 39, KDE
Keyboard: M01, Default EN-US Layout
Chrysalis Version: 0.13.3
Firmware Version: 0.92.1+94

On initial glance it appears that they have shifted by one row, but some of the moved keys don't  follow this pattern.

obra, 2024-01-17:
@ahusby I'm trying to replicate this, (although I'm on macOS) and with an Atreus with the same setup (including everything set to Norwegian Bokmål and am unable to replicate what you're seeing.  I'd really like to make sure it's not something in the Gnome settings. Is it possible for you to test on another host or even at the (non-graphical) console?

ahusby, 2024-01-18:
@obra Doh! When you mentioned Gnome settings I realized I have indeed been messing with my Gnome settings. A long time ago, in the app "Gnome Tweaks", I have enabled the option to swap left alt with left ctrl. I forgot. Sorry, I appologize for wasting your time.

I have very little time to do further testing to verify this, unfourtunately. I'm okay with closing this Github issue.

That setting was the cheapest, easiest way to improve my typing experience on my old, traditional, row staggered keyboard. Since then I have bought the Atreus, which I hope will become my daily driver, thus removing the need for that setting.

[image]

obra, 2024-01-18:
I'm happy to hear that it wasn't an issue with Chrysalis or Kaleidoscope!

closed 2024-01-18 by obra