keyboardio/Chrysalis #1324: chrysalis.keyboard.io spinning forever

pheuberger, 2024-02-28:
**Describe the bug**
My Chrysalis software on the laptop didn't work for a while now so I thought I might have more luck with the web version, but that doesn't seem to work either. After selecting the device in the popup I now see the spinner with the text "Reading data from device..." and nothing happens. It's been minutes already.

**Expected behavior**
I expect to be able to see my keyboard with its current layout and be able to make changes to it.

**Desktop (please complete the following information):**
 - OS: Fedora 39
 - Browser: Chrome Version 121.0.6167.160 (Official Build) (64-bit)
 - Chrysalis Version: web

obra, 2024-02-28:
Hi,

I suspect that the issue you're running into is with corrupted data in your
keyboard's storage, but I don't know for sure. Would it be possible to grab
the log of the browser console after you attempt a connection?
Alternatively, on desktop Chrysalis, grab a debugging bundle and attach it
to this ticket.  (Also, what keyboard are you using?)

pheuberger, 2024-03-01:
Ah, sorry, completely forgot to include the keyboard model haha. I have the Model 100.
Here's all the output from the console:
```
[2024-03-01T13:42:26.604Z] [DEBUG] [a.navigate (https://chrysalis.keyboard.io/js/main.23d496b30b9c13ace8f2.bundle.js:205:666579)]: navigating to location Objectargs: ['/keyboard-select'][[Prototype]]: Objectconstructor: ƒ Object()hasOwnProperty: ƒ hasOwnProperty()isPrototypeOf: ƒ isPrototypeOf()propertyIsEnumerable: ƒ propertyIsEnumerable()toLocaleString: ƒ toLocaleString()toString: ƒ toString()valueOf: ƒ valueOf()__defineGetter__: ƒ __defineGetter__()__defineSetter__: ƒ __defineSetter__()__lookupGetter__: ƒ __lookupGetter__()__lookupSetter__: ƒ __lookupSetter__()__proto__: (...)get __proto__: ƒ __proto__()set __proto__: ƒ __proto__()
main.23d496b30b9c13ace8f2.bundle.js:205 [2024-03-01T13:42:26.660Z] [LOG] [En (https://chrysalis.keyboard.io/js/main.23d496b30b9c13ace8f2.bundle.js:205:1446921)]: [object Object]
main.23d496b30b9c13ace8f2.bundle.js:205 [2024-03-01T13:42:26.661Z] [LOG] [En (https://chrysalis.keyboard.io/js/main.23d496b30b9c13ace8f2.bundle.js:205:1446921)]: null
main.23d496b30b9c13ace8f2.bundle.js:205 [2024-03-01T13:42:26.690Z] [DEBUG] [a.navigate (https://chrysalis.keyboard.io/js/main.23d496b30b9c13ace8f2.bundle.js:205:666579)]: navigating to location Object
main.23d496b30b9c13ace8f2.bundle.js:205 [2024-03-01T13:42:26.725Z] [LOG] [En (https://chrysalis.keyboard.io/js/main.23d496b30b9c13ace8f2.bundle.js:205:1446921)]: [object Object]
main.23d496b30b9c13ace8f2.bundle.js:205 [2024-03-01T13:42:26.726Z] [LOG] [En (https://chrysalis.keyboard.io/js/main.23d496b30b9c13ace8f2.bundle.js:205:1446921)]: null
main.23d496b30b9c13ace8f2.bundle.js:205 [2024-03-01T13:42:30.446Z] [LOG] [u (https://chrysalis.keyboard.io/js/main.23d496b30b9c13ace8f2.bundle.js:205:882770)]: in connectToKeyboard
main.23d496b30b9c13ace8f2.bundle.js:205 [2024-03-01T13:42:30.451Z] [LOG] [En (https://chrysalis.keyboard.io/js/main.23d496b30b9c13ace8f2.bundle.js:205:1446921)]: [object Object]
main.23d496b30b9c13ace8f2.bundle.js:205 [2024-03-01T13:42:30.451Z] [LOG] [En (https://chrysalis.keyboard.io/js/main.23d496b30b9c13ace8f2.bundle.js:205:1446921)]: null
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=00001800-0000-1000-8000-00805f9b34fb, name='MX Ergo'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=00001801-0000-1000-8000-00805f9b34fb, name='MX Ergo'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000180a-0000-1000-8000-00805f9b34fb, name='MX Ergo'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000180f-0000-1000-8000-00805f9b34fb, name='MX Ergo'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=00001812-0000-1000-8000-00805f9b34fb, name='MX Ergo'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=00001108-0000-1000-8000-00805f9b34fb, name='iBose45'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000110b-0000-1000-8000-00805f9b34fb, name='iBose45'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000110c-0000-1000-8000-00805f9b34fb, name='iBose45'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000110d-0000-1000-8000-00805f9b34fb, name='iBose45'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000110e-0000-1000-8000-00805f9b34fb, name='iBose45'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000111e-0000-1000-8000-00805f9b34fb, name='iBose45'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000112f-0000-1000-8000-00805f9b34fb, name='iBose45'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=00001200-0000-1000-8000-00805f9b34fb, name='iBose45'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=00001000-0000-1000-8000-00805f9b34fb, name='AirPods - Find My'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000110b-0000-1000-8000-00805f9b34fb, name='AirPods - Find My'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000110c-0000-1000-8000-00805f9b34fb, name='AirPods - Find My'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000110e-0000-1000-8000-00805f9b34fb, name='AirPods - Find My'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000111e-0000-1000-8000-00805f9b34fb, name='AirPods - Find My'
[... cut ...]  80 81 82 79 65535 65535 77 70 73 65535 20548 20518 65535 65535 65535 65535 18658 18666 18665 65535 49 2097 
main.23d496b30b9c13ace8f2.bundle.js:205 [2024-03-01T13:42:36.471Z] [DEBUG] [e.value (https://chrysalis.keyboard.io/js/main.23d496b30b9c13ace8f2.bundle.js:205:406839)]: request Object
main.23d496b30b9c13ace8f2.bundle.js:205 [2024-03-01T13:42:36.721Z] [DEBUG] [e.<anonymous> (https://chrysalis.keyboard.io/js/main.23d496b30b9c13ace8f2.bundle.js:205:408135)]: Making a request keymap.custom
```

tomako, 2024-03-01:
I'm in exactly the same boat but on Windows (Model100).
Old binary version stopped working 6+ months ago and I jumped on the web version with full of optimism.
As I see the log on the console is also the same.

obra, 2024-03-01:
@pheuberger Ok. Given the firmware version you're running, I suspect that the issue is indeed data corruption, either in the stored copy of the keymaps or in the wire protocol.

If you're comfortable with a factory reset (and losing your custom keymaps), can you try holding in the 'prog' key while connecting your keyboard to the computer, so that the led turns red to indicate it's in bootloader mode, then connecting from https://chrysalis.keyboard.io and doing a Firmware Update with 'factory reset' selected?

pheuberger, 2024-03-03:
I had to download Chrysalis to do the factory reset, because it didn't recognize the device in bootloader mode. See screenshot below. Anyways, it worked in the app, so thanks a lot! :pray:

closed 2024-03-03 by pheuberger