keyboardio/Chrysalis #1325: Atreus Connection Issues On `chrysalis.keyboard.io`

nlordell, 2024-02-29:
When trying to connect my Atreus keyboard on [chrysalis.keyboard.io](https://chrysalis.keyboard.io), I get the following message:

> I couldn't connect to the keyboard. It may already be connected in another application or browser window. The error I got was: NetworkError: Failed to execute 'open' on 'SerialPort': Failed to open serial port.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to <https://chrysalis.keyboard.io>
2. Click on 'Connect'
3. See error

**Expected behavior**
Connect works.

**Desktop (please complete the following information):**
 - OS: Fedora 39 + Chromium 122
 - Chrysalis Version: chrysalis.keyboard.io

<details><summary><b>Report Bundle</b></summary>

```json
{
  "logs": [
    {"message": "[2024-02-29T12:19:19.024Z] [DEBUG] [t.navigate (https://chrysalis.keyboard.io/js/main.0003b1c84532178a6aa3.bundle.js:205:666536)]: navigating to location", "data": {"args": ["/keyboard-select"]}},
    {"message": "[2024-02-29T12:19:19.034Z] [DEBUG] [t.navigate (https://chrysalis.keyboard.io/js/main.0003b1c84532178a6aa3.bundle.js:205:666536)]: navigating to location", "data": {"args": ["/keyboard-select"]}},
    "[2024-02-29T12:19:19.057Z] [LOG] [kn (https://chrysalis.keyboard.io/js/main.0003b1c84532178a6aa3.bundle.js:205:1462261)]: [object Object]",
    "[2024-02-29T12:19:19.057Z] [LOG] [kn (https://chrysalis.keyboard.io/js/main.0003b1c84532178a6aa3.bundle.js:205:1462261)]: null",
    "[2024-02-29T12:19:21.092Z] [LOG] [u (https://chrysalis.keyboard.io/js/main.0003b1c84532178a6aa3.bundle.js:205:881534)]: in connectToKeyboard",
    "[2024-02-29T12:19:21.094Z] [LOG] [kn (https://chrysalis.keyboard.io/js/main.0003b1c84532178a6aa3.bundle.js:205:1462261)]: [object Object]",
    "[2024-02-29T12:19:21.095Z] [LOG] [kn (https://chrysalis.keyboard.io/js/main.0003b1c84532178a6aa3.bundle.js:205:1462261)]: null",
    {"message": "[2024-02-29T12:19:23.739Z] [ERROR] [https://chrysalis.keyboard.io/js/main.0003b1c84532178a6aa3.bundle.js:205:889261]: error while trying to connect", "data": {"error": {"stack": "Error: Failed to execute 'open' on 'SerialPort': Failed to open serial port."}, "device": null}},
    {"message": "[2024-02-29T12:19:23.740Z] [DEBUG] [https://chrysalis.keyboard.io/js/main.0003b1c84532178a6aa3.bundle.js:205:631312]: Toast message received", "data": {"msg": {"variant": "error", "message": "I couldn't connect to the keyboard. It may already be connected in another application or browser window. The error I got was: NetworkError: Failed to execute 'open' on 'SerialPort': Failed to open serial port."}, "label": "toast"}},
    "[2024-02-29T12:19:23.743Z] [LOG] [kn (https://chrysalis.keyboard.io/js/main.0003b1c84532178a6aa3.bundle.js:205:1462261)]: [object Object]",
    "[2024-02-29T12:19:23.743Z] [LOG] [kn (https://chrysalis.keyboard.io/js/main.0003b1c84532178a6aa3.bundle.js:205:1462261)]: null"
  ]
}
```

</details>

**Additional Context**
One thing I noticed is that, when connecting from a newly opened Chromium window and I see the WebUSB dialog pop up, I see the the 'Atreus (tty...)' entry. When I select it, I get the connection error as described :point_up:. If I try to connect again, I see an entry 'Atreus (tty...) - Paired' in the dialog instead, suggesting that the first connection attempt is doing _something_. Connection fails the second time as well:

[image]
[image]

Also note that this is a fairly fresh Fedora install, I vaguely remember having to install udev rules in the past for my Model01, and wonder if it may be related to that?

obra, 2024-03-01:
> entry. When I select it, I get the connection error as described ☝️. If I

I think the issue is that udev isn't configured to let you connect. I've
just pushed a Chrysalis update. I'd love to know if it works for you. (I
expect that it will give you a much better error message that tells you how
you need to solve the problem)

nlordell, 2024-03-01:
I did some debugging and got it to work, a few comments:

- Yes, the text was super helpful in debugging. If I hadn't previously had to mess with `udev` rules when setting up my Model01 keyboard in the past, I would have likely taken a lot longer to figure out where to start looking
- There is a `user-select: none` CSS property on the help which is annoying, in particular you cannot copy the path to the `rules.d` directory or the `.rules` file name. Note that the rule doesn't apply to the `udev` rules themselves, so those can be copied.
- I tried the `udev` rules from the help text but they didn't work for me. Note that it works using the [`60-kaleidoscope.rules`](https://github.com/keyboardio/Kaleidoscope/blob/master/etc/60-kaleidoscope.rules) rules from the Kaleidoscope repository. I don't know if it is the ModemManager issue or the `seat` tag is missing (my understanding of `udev` is super limited).

So, for others having the same issue, I suggest following the [instructions](https://github.com/keyboardio/Kaleidoscope/blob/master/docs/setup_toolchain.md#install-arduino-on-linux) from Kaleidoscope for now.

obra, 2024-03-01:
I really appreciate the feedback!

I've just pushed an updated version of Chrysalis that should correct the
rules and the user-select.

closed 2024-03-05 by obra

ewhd, 2024-05-13:
@obra 

Not sure if I should create a new issue or revive this one, I experienced the same problem as nlordell, and thanks to this thread solved it by downloading [60-kaleidoscope.rules](https://github.com/keyboardio/Kaleidoscope/blob/master/etc/60-kaleidoscope.rules).

After some testing I think I was able to narrow down the problem to a syntax difference:

The udev rules the error message instructed me to save as `50-keleidoscope.rules` have lines that look like:
```
SUBSYSTEM=="usb", ATTR{idVendor}=="3496", ATTR{idProduct}=="0005"
```

Whereas the udev rules in `60-kaleidoscope.rules` look like:
```
SUBSYSTEMS=="usb", ATTRS{idVendor}=="3496", ATTRS{idProduct}=="0005"
```
With `SUBSYSTEMS` and `ATTRS` now having "S" on the end.

When I changed those terms on each line in `50-kaleidoscope.rules` to match, it works.

Oh, and also I have to unplug and re-plug the keyboard when I run `sudo udevadm control --reload-rules` to help the rules take effect, a step included in [these instructions](https://github.com/keyboardio/Kaleidoscope/blob/master/docs/setup_toolchain.md#install-arduino-on-linux) but which the error message did not mention.

obra, 2024-05-13:
Thank you! That's the mistake I was looking for and not spotting. I'll get
that fixed today.

obra, 2024-05-13:
Chrysalis 2024.0513.2048 contains this fix.