keyboardio/Chrysalis #1330: chrysalis.keyboard.io firmware update can't find device during "Entering programmable mode"

alunduil, 2024-03-20:
**Describe the bug**
During the "Entering programmable mode" step of "Firmware Update," a dialogue titled "Connecting to your keyboard..." appears allowing me to click "OK" to get a device list. The resulting device list is empty on my system.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Firmware Update'
2. Click on 'UPDATE'
3. Hold "prog" key
4. Click "Continue"
5. "Connecting to your keyboard..." dialogue appears
6. Click "OK"
7. Browser dialogue appears with "No compatible devices found."

**Expected behavior**
Firmware update successfully completes.

**Screenshots**
[image]
[image]
BUNDLE: https://gist.github.com/alunduil/10f4d556ff53460d05eca932bf6f74a7

**Desktop (please complete the following information):**
 - OS: Google ChromeOS Version 122.0.6261.118 (Official Build) (64-bit)
 - Chrysalis Version: Chrysalis 2024.0318.1913

**Additional context**
This is the first firmware update I'm attempting on this Keyboardio Model 100.

obra, 2024-03-20:
Hi Alex,

I don't have a current ChromeOS device here to test, but I've got some
guesses.

At the point where the empty device chooser pops up, is the prog key red?
If not, try disconnecting the keyboard and holding in the 'prog' key while
connecting it. Does it show up then?

If not:

 If you hold down the 'prog' key while plugging the keyboard into the
Chromebook, the Prog key should turn red. If you then load up Chrysalis and
try to connect, it will show an empty device chooser. Click "cancel" then.
Does the Model 100 show up when the second device chooser pops up?  (There
are some web platform limitations that don't let us probe your keyboard to
be able to show you only the correct device chooser.)

Best,
jesse

alunduil, 2024-03-23:
Thanks, the following is how I got my firmware updated in case it's useful to others.

1. Browse to chrysalis.keyboard.io
2. Click "CONNECT"
3. Select "Model 100"
4. Navigate to "Firmware Update"
5. Click "UPDATE"
6. Hold "prog"
7. Click "CONTINUE"
8. At the next device prompt
9. Disconnect keyboard
10. Hold "prog" while reconnecting keyboard
11. Select bootable Model 100
12. Continue with firmware update as normal

Thanks for the pointers @obra

obra, 2024-03-25:
Thanks so much for the update, Alex. I just created #1333 to track adding
better in-ui docs for this situation.

nblumoe, 2024-04-17:
Thanks, the procedure described by @alunduil also worked with my Model01.
For the record: It was always shown correctly in the device picker, but it never starting flashing.

This is on macOS 14.4.1

iGEL, 2024-10-18:
Just experienced the same bug on Linux Mint using Chromium. The instructions @alunduil shared in his 2nd post (disconnect + hold prog when reconnecting) helped. 👍

closed 2024-11-11 by obra