keyboardio/Chrysalis #1335: Original Model01 keyboard with original firmware can't use Chrysalis

belden, 2024-04-02:
**Describe the bug**

This is a continuation of https://github.com/keyboardio/Chrysalis/issues/1034.

> _Chrysalis now runs in-browser using WebSerial and WebUSB and this functionality has been rewritten, so I'm closing out this issue as obsolete. Please don't hesitate to open a new issue if https://chrysalis.keyboard.io exhibits the same behavior_
>
> _Originally posted by @obra in https://github.com/keyboardio/Chrysalis/issues/1034#issuecomment-1965248724_

**To Reproduce**
Steps to reproduce the behavior:
1. Attach a Model01 which has never had a firmware update.
2. Visit https://chrysalis.keyboard.io/.
3. Scroll to the bottom and click `[Connect]`

**Expected behavior**

The configuration tool should successfully read data from my keyboard and present a configuration UI.

What actually happens is that the `[Connect]` button is replaced with a loading spinner which never resolves. Various errors are logged to the browser console.

**Screenshots**

n/a

**Desktop (please complete the following information):**
 - OS: [OSX Sonoma 14.4]
 - Chrysalis Version: [2024.0318.1913]

**Additional context**

See also https://github.com/keyboardio/Chrysalis/issues/1034

My underlying goal is really simple; I keep accidentally bumping `Fn-n`, which acts as mute 🔇. I want to just turn that into a volume-down control. This was an issue for me ~2 years ago when I opened issue 1034, then stopped being an issue when I switched from OSX to Linux. I'm back on OSX again, so here I am with my same old Model01, trying to disable the mute switch! 🤣

belden, 2024-04-02:
I wasn't able to create the debug-bundle as described in the bug-issue template. Here is a copy of the javascript console as suggested as a fall-back.

[chrysalis.keyboard.io-1712089350852.log](https://github.com/keyboardio/Chrysalis/files/14843417/chrysalis.keyboard.io-1712089350852.log)

chongfun, 2024-04-10:
I had the same issue but resolved it by flashing the firmware using 0.8.6.

ChristianHarnisch, 2024-09-07:
I have the same issue. My OS is Windows 10 and Chrysalis is Version 2024.0830.2138.

The little spinner on the "CONNECT" button never stops. Clicking "Layout Card" or "Backup and Restore layouts" in the sidebar leads to a screen with a rotating flower/mandala and the text "Reading data from device..." below it - and nothing else happens.

I suppose an old release of Chrysalis (like the aforementioned 0.8.6) can be used as a workaround: On startup, it offers to update the firmware. However, I haven't updated yet - @obra is this safe, or should I wait for a fix?

obra, 2024-09-09:
@ChristianHarnisch What version of Kaleidoscope is running on your keyboard?

Can you click on the sidebar menu and get the debugging bundle from the 'report an issue' screen?

ChristianHarnisch, 2024-09-09:
@obra Thanks for your reply.

I can access the "report an issue" screen, but clicking "CREATE BUNDLE" does nothing (except for playing a short animation on the button itself).

obra, 2024-09-09:
It's going to take a while if it's having trouble talking to the keyboard. Alternatively, open up the browser's devtools and grab the full console log after attempting a connection to the keyboard.

ChristianHarnisch, 2024-09-09:
Sorry, that was apparently my fault - clicking "CONNECT" first leads "CREATE BUNDLE" to do nothing. Without that, this is the output that appears (pretty-printed for readability):

```
{
  "logs": [
    "[2024-09-09T20:35:06.727Z] [DEBUG] [u.resetDeviceState (https://chrysalis.keyboard.io/js/main.ae3dd3236fc33ec3893c.bundle.js:205:353612)]: Resetting device state",
    {
      "message": "[2024-09-09T20:35:06.801Z] [DEBUG] [https://chrysalis.keyboard.io/js/main.ae3dd3236fc33ec3893c.bundle.js:205:484490]: navigating to location",
      "data": {
        "args": [
          "/keyboard-select"
        ]
      }
    },
    "[2024-09-09T20:35:06.855Z] [LOG] [kn (https://chrysalis.keyboard.io/js/main.ae3dd3236fc33ec3893c.bundle.js:205:967735)]: [object Object]",
    "[2024-09-09T20:35:06.855Z] [LOG] [kn (https://chrysalis.keyboard.io/js/main.ae3dd3236fc33ec3893c.bundle.js:205:967735)]: null"
  ]
}
```

obra, 2024-09-10:
Oh hang on. The *original* firmware for the Model 01? Right. The original
Model 01 predated the existence of Chrysalis. You need to do a firmware
update and factory reset before anything else will work.

ChristianHarnisch, 2024-09-10:
No worries, doing a firmware update etc. would be fine by me. But how do I do that?

If I understand [this comment by algernon](https://github.com/keyboardio/Chrysalis/issues/1034#issuecomment-1272298797) (and the others preceding it) correctly, Chrysalis did at some point tell how to perform the update. But that doesn't happen with the new web-based one.

You could (re-)implement that or I could temporarily use an old release. Perhaps the former wouldn't be worth the effort, because how many original firmware Model 01s are still out there realistically? On the other hand, the latter depends on if the "intermediate page" and "firmware update screen" still work - URLs and download locations might have changed etc. That's what I meant when I asked "is this safe", and I'm sorry for not wording that clearly.

What do you think I should do?

obra, 2024-09-11:
If you get to the browser's 'pick a device' widget on web-based Chrysalis,
disconnect the Model 01 from the computer, then while holding in 'Prog',
connect it. You should see Model 01 Bootloader as an option. From there, it
should push you into the firmware update workflow.

ChristianHarnisch, 2024-09-16:
> If you get to the browser's 'pick a device' widget on web-based Chrysalis, disconnect the Model 01 from the computer, then while holding in 'Prog', connect it. You should see Model 01 Bootloader as an option. From there, it should push you into the firmware update workflow.

Thank you so much! That did the trick.

One detail I noticed: I had to keep "prog" pressed the entire time from connecting the keyboard until starting the firmware update. Not just again before clicking the confirmation dialog (as I understood its instructions initially). Otherwise I'd get an unspecific "failed to update firmware" message.

In any case, again, thanks for your help.

obra, 2024-09-17:
Glad it's behaving! And indeed, I need to redo the text and flow around
button holding to make that clearer.

closed 2024-11-11 by obra