keyboardio/Chrysalis #1337: Unable to toggle "Sticky" key modifier for Model 01

tobanw, 2024-04-15:
**Describe the bug**
I've updated my Model 01 to the latest firmware (including factory reset) on https://chrysalis.keyboard.io/. Now I'm using the layout editor but I'm unable to toggle the "Sticky" modifier for any key -- it's grayed out, and the hovertext does not say it's unsupported by the firmware (as it does for other grayed-out buttons).

**To Reproduce**
Steps to reproduce the behavior:
1. Connect Model 01 on https://chrysalis.keyboard.io/ and update firmware if needed
2. In the Layout & Colormap Editor, select the desired modifier key on the keyboard, then go to the "Modifiers" tab in the bottom panel
3. See that the "Sticky" toggle is grayed-out regardless of which key is selected

**Expected behavior**
"Sticky" toggle is operational.

**Screenshots**
In this screencap, the cursor is hovering the "Sticky" toggle, so you can see the hovertext.
[image]

**Desktop (please complete the following information):**
 - OS: macOS Sonoma 14.4.1
 - Chrysalis Version: 2024.0318.1913

**Additional context**
Previously, I had been using QMK to configure layouts, but I figure the firmware update and reset would wipe the slate clean.

obra, 2024-04-17:
The Model 01's available flash memory is pretty limited, so the default
firmware doesn't include all the features we might like to have in there.

If you're looking for this functionality on the Model 01, you'll need to
flash a custom firmware build that removes some of the LED effects and
enables the OneShot suite of plugins. (It should be the case that if you
hover over the broader category, it gives you a warning about your firmware
not having support for this feature)

-jesse

tobanw, 2024-04-21:
Thanks! It looks like I'll need to modify the `.ino` file here to customize the build: https://github.com/keyboardio/Model01-Firmware

Is it enough to add `#include <Kaleidoscope-OneShot.h>` to the `.ino` file? If not, can you point me to the docs?

obra, 2024-04-22:
You'll also need to add it to the plugin list.
https://kaleidoscope.readthedocs.io/en/latest/plugins/Kaleidoscope-OneShot.html

tobanw, 2024-05-01:
I got it working! But not as instructed in the docs: if I enable the `OneShotConfig` plugin in the firmware `.ino` file, the keys gets all garbled with keys seemingly offset by some fixed amount (e.g., `7890` became `qwer` and `uiop` became `asdf`).

Now I have it working as desired with both `OneShot` and `EscapeOneShot` plugins enabled, but not `OneShotConfig` nor `EscapeOneShotConfig`. I'm still able to use Chrysalis to toggle "Sticky", so it seems that `OneShotConfig` is not needed. (I also see I can use the "sticky cancel" key so `EscapeOneShotConfig` also appears to be redundant.) Though I'll note that when saving the layout in Chrysalis, it hung on the saving spinner, but the changes did take effect. Had to unplug and reconnect my Model-01 to get Chrysalis to connect to it after that.

Let me know if this is a known issue -- if not, I can file a separate issue for this.

obra, 2024-05-02:
That sounds like the issue was that you flashed from the commandline with a
config in eeprom already. It should be the case that if you do a "Factory
reset" from Chrysalis, it'd get the corrupted layout cleared out and you'd
still be able to use those config plugins.

best,
jesse

closed 2024-11-11 by obra