keyboardio/Chrysalis #1339: "firmware limitations" info bubble obscures the Layer pop-up menu.

parke, 2024-05-03:
**Describe the bug**
On the `Modifiers` tab, if the secondary action is set to `Layer shift`, then an info bubble appears with the following message:

> Due to firmware limitations, shifting to layers above "<layer_name>" is not supported via secondary actions.

Then, if the `Layer` pop-up menu is opened, the above info bubble text appears on top of the pop-up menu and obscures part of the menu.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the `Modifiers` tab.
2. Set `When held` to `Layer shift`
3. Hover over the `Layer` pop-up menu
4. Note the appearance of the info bubble.
5. Click on the `Layer` select box to open the pop-up menu.
6. Note that the info bubble is on top of the pop-up menu, obscuring some of the items in the menu.

**Expected behavior**
I would expect to be able to see the entire pop-up menu.

One option might be to move the text to a tool-tip that would only appear when the mouse hovers over a greyed-out layer in the pop-up menu.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Linux
 - Chrysalis Version: 2024.0501.1931

TrueFalcon, 2024-05-04:
I found another. If you hover over the top of the Layer menu at top right where it says "Layer     Key#0", a popup prevents you from selecting Layer 0 or 1 until you move away and let the popup disappear. Oops! One more: Under Modifiers, hover over "Invert Shift" and "Sticky" is obscured. I couldn't check under Leds since my Atreus doesn't have any! 😉

obra, 2024-05-15:
I think that some updates I just pushed to how our tooltips work *should*
take care of this. Please reply or reopen this issue if it's still
misbehaving for you.

closed 2024-05-15 by obra