keyboardio/Chrysalis #1340: Modifier keys get stuck

min-pascal, 2024-05-11:
After updating to v0.92.6+116 modifier keys occasionally get stuck. Pressing them again gets them up again

obra, 2024-05-13:
This sounds like there may be an EEPROM corruption issue on your keyboard,
or there may be a sticky keys setting you need to toggle.

Can you attach the results of 'Report a problem' in the Chrysalis menus
after connecting your keyboard?

min-pascal, 2024-05-14:
Okay. I'm guessing it is a corrupted EEPROM then. My layouts got screwed after the update and I have had random keys light up in random colors. Do I need to factory reset the Keyboard?

obra, 2024-05-14:
That's likely the fastest way to get you up and running again.

best,
jesse

closed 2024-05-15 by obra

min-pascal, 2024-05-16:
I have reset the keyboard to factory resets with chrysalis but there are 2 things that confuse me. The firmware version 0.92.6+116 was still written in the UI after the reset and there is still misalignment with the key colors given in the software and the keys who have color.

obra, 2024-05-17:
> I have reset the keyboard to factory resets with chrysalis but there are 2
> things that confuse me. The firmware version 0.92.6+116 was still written
> in the UI after the reset and there is still misalignment with the key
> colors given in the software and the keys who have color.
The firmware version doesn't get reset with a factory reset, just the data
stored in the keyboard's persistent storage. Are your modifier keys now
behaving?   With a straight up factory reset, there shouldn't have been
*any* hardcoded key colors. Can you tell me a bit more about what you're
seeing in detail?

min-pascal, 2024-05-19:
I think I've narrowed down what's going on with the inconsistent colouring of keys. All keys on layer #1 insist to be red if they are not transparent. I moved my layer #1 to another layer as a workaround to get them green. Despite from that the keyboard is functioning correctly. No more sticky modifiers. I do on the other hand experience an occasional bug in chrysalis where keys suddenly change to a 6 digit hex number when editing but that is another and luckily not a breaking thing

obra, 2024-05-20:
Oh. So this is a thing that will change soon, but the second layer (Layer
1) is, by default, the Numpad layer. And since the first release of the
Model 01, we've had a hardcoded effect that lights the Numpad up as red. In
the next firmware release or so, we're going to be moving to a more modern,
flexible system, where the LED effect uses a more customizable LED overlay
that you'll be able to configure.

Right now, changing what's going on with layer 1 requires compiling a new
firmware "sketch" for your keyboard. Alternatively, you could skip layer 1
with your custom config and use higher layers instead.