keyboardio/Chrysalis #1343: Updating the firmware on Atreus can hang indefinitely.

NabiYang, 2024-08-08:
**Describe the bug**
Flashing the firmware on Atreus can get stuck on step 3 Flashing. The developer console says that the device was lost. However, after successfully updating my firmware using macOs and Fedora Linux, I am now able to successfully update my firmware in Windows, so I can no longer reproduce this issue. Nonetheless, I investigated it because another user reported in Discord that they were unable to update in Windows, so this may still be an issue for others.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://chrysalis.keyboard.io/
2. Connect and attempt to update the firmware.
3. It freezes on Step 3 and never completes.

**Expected behavior**
I expect a green banner to appear informing me that the firmware has updated successfully after it finishes all steps.

**Screenshots**
[image]
[bundle.txt](https://github.com/user-attachments/files/16551575/bundle.txt)

**Desktop (please complete the following information):**
 - OS: Windows 11 Pro 23H2
 - Chrysalis Version: Chrysalis 2024.0518.0129

**Additional context**
When I was unable to update through the web app, I was able to use the desktop app to update, however this didn't make the web app work after. The web app in Windows only started working after I tried updating on macOs and Linux.

maik-dobryn-sgd, 2024-08-20:
Same issue here...

obra, 2024-08-20:
@maik-dobryn-sgd - Could you please include the bundle.txt from 'report an issue' sidebar menu after you try to update?

cattmampbell, 2024-08-21:
The same thing happens to me. @obra, I'm not 100% sure which `bundle.txt` file you are referencing (I couldn't find such a file in the Chrome Developer Tools). However I did take save the output from the Console (`chrysalis.keyboard.io-1724258244348.log`), save a copy of `bundle.js` (renamed to `main.b89e52652476f8656928.bundle.js.txt`), and take a few screenshots.

While Flashing the latest Chrysalis Firmware, I saw _this_ in the Console:

[image]

Clicking on the `https://chrysalis.keyboard.io/js/main.b89e52652476f8656928.bundle.js:205:335678` hyperlink in the last Console message led me here:

[image]

Clicking on the `main.b89e526... .bundle.js:205` hyperlink in the last Console message led me here:

[image]

Here are the aforementioned output from the Console (`chrysalis.keyboard.io-1724258244348.log`) and copy of `bundle.js` (renamed to `main.b89e52652476f8656928.bundle.js.txt`). 

[chrysalis.keyboard.io-1724258244348.log](https://github.com/user-attachments/files/16695461/chrysalis.keyboard.io-1724258244348.log)

[main.b89e52652476f8656928.bundle.js.txt](https://github.com/user-attachments/files/16695481/main.b89e52652476f8656928.bundle.js.txt)

I hope this helps narrow down this bug.

I have owned my Keyboardio Atreus for a couple of years now, but I have never used Chrysalis before. I only attempted to update the Chrysalis firmware in order to employ the Secondary Action feature.

maik-dobryn-sgd, 2024-08-22:
> @maik-dobryn-sgd - Could you please include the bundle.txt from 'report an issue' sidebar menu after you try to update?

[bundle.txt](https://github.com/user-attachments/files/16718982/bundle.txt)

obra, 2024-08-30:
Ok. It took some spleunking, but it looks like what was going on was that windows was sometimes splitting the message from the keyboard that it was in bootloader mode into two separate packets, so instead of one message reading "CATERIN", it was getting a message reading "C" and another reading "ATERIN" ... but only sometimes.

I've just pushed an update to Chrysalis that should resolve this issue.

closed 2024-08-30 by obra