keyboardio/Chrysalis #1345: how to improve the response of the palm key on Model 100

qliu-Inscopix, 2024-08-14:
**Describe the bug**

Model 100:

My left palm key, set as a layer shift key, sometimes responds slowly when used with other keys, causing unintended solo activation of those keys. How can I fix this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem. If possible, please also include the debug bundle you can create on the "Report a problem" screen. If, for some reason, you are unable to access that screen, please press `Control + Shift + I` (Windows or Linux) or `Command + Option + I` (macOS) to bring up the developer console. Right-clicking inside it will allow you to save its contents to a file - please attach that to the bug report.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Chrysalis Version: [e.g. 0.4.0]

**Additional context**
Add any other context about the problem here.

obra, 2024-08-14:
It shouldn’t be possible for it to respond *slowly* although you may not be
getting it pressed all the way to the actuation point before hitting those
other keys. One possible solution would be to switch your palm keys to
kailh “speed” switches, which actuate much earlier in their keystrokes.

Jesse

qliu-Inscopix, 2024-08-15:
Thank you. Speed switches sound promising.

closed 2024-08-30 by obra