keyboardio/Chrysalis #1346: Error on installing imported JSON layout

oscar-schwarz, 2024-08-15:
**Describe the bug**
I exported a [layout](https://github.com/user-attachments/files/16628254/Chrysalis_Keyboardio-Atreus_layout_2024-08-15_19-19-02.json) from the web version of the app and imported it into the desktop version. After trying to save the layout I get `
TypeError: Cannot read properties of undefined (reading 'length')`
After reconnecting to the keyboard the layout is scrambled on some layers.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect a Keyboardio Atreus with the latest firmware
2. Click on "Backup and Restore"
3. Click on "Load from File..." and select the JSON from above
4. Click on "Import"
5. In the Layout Editor try to save
6. Now the error should be displayed

**Expected behavior**
No error occurs and layout gets saved correctly.

**Screenshots**
[chrysalis-debug.json.txt](https://github.com/user-attachments/files/16628449/chrysalis-debug.json.txt)

**Desktop (please complete the following information):**
 - OS: Linux
 - Chrysalis Version: 0.13.3

obra, 2024-08-15:
Unfortunately, the web based version isn't backwards compatible with the
old desktop version.

oscar-schwarz, 2024-08-15:
I see. The only reason I tried the desktop version was because on the web version the import fails too. I will edit this report later so it resembles the web bug.

obra, 2024-08-15:
Sure thing. That should absolutely work.    The debug log from the web import attempt would be very helpful.

Best,
Jesse

oscar-schwarz, 2024-08-16:
I updated the issue so that it represents the web bug.

oscar-schwarz, 2024-08-16:
I managed to repair my layout json by comparing it to a similar working file export. Heres the [diff](https://github.com/user-attachments/files/16636151/repair_layout.txt). It seems the structure of the last layer was messed up. But that it the applications fault. I havent messed with that file. But I have no idea how to reproduce that wrong export behaviour.

obra, 2024-08-16:
That looks like it was missing a layer. Do you happen to know what firmware
revision you were coming from?

oscar-schwarz, 2024-08-16:
Sadly not. But I know it was a version without mouse and macro support and layer 8 and 9 were greyed out in Chrysalis.

obra, 2024-08-16:
Interesting! I don't think we ever shipped an 'official' firmware without
mouse support. But that gives me an idea of what might be causing the
problem.  I'll retitle this bug, but it sounds like Chrysalis isn't robust
enough when loading keymaps that don't have enough layers.