keyboardio/Chrysalis #1355: Please mention the Japanese 英数 and かな keys where appropriate.

charles-plessy, 2024-10-13:
Hello,

thank you so much for developping and selling your keyboards.  My order just arrived today and it is a wonderful piece of hardware.

To use it as Japanese keyboard with a Macintosh, I made a bit of research.  It took a lot of time but the answer is very simple. Maybe you can display it somewhere in the documentation or the user interface, for instance if the user selects Japanese keyboard layout in the preferences?  Here is the key information:

 - The `japanese_eisuu` key (英数) is Lang2, key code 145.
 - The `japanese_kana` key (かな) is Lang1, key code 144.

That's all!  With just this information in hand Japanese users can configure their Keyboardio to place these keys as they please and enjoy these wonderful keyboards on their Macs.

Have a nice day,

Charles

charles-plessy, 2024-10-16:
Both can be true: Muhenkan is on the PC keyboards (which have more special Japanese keys). Eisu/Kana are on the Mac Japanese keyboards (which only have these two).

obra, 2024-10-16:
Can you point me to a reference showing Eisu as being mapped to Lang2? I
couldn’t dig one up today.

charles-plessy, 2024-10-16:
I also could not find online documentation but I checked directly the keycodes on my Japanese Mac with https://karabiner-elements.pqrs.org/.

charles-plessy, 2024-10-16:
Oh, sorry there was. https://bsakatu.net/doc/usb-hid-to-scancode/

obra, 2024-10-17:
Yep. I’ll update the UI. I’ve found more resources confirming this. Do you happen to have access to a PC keyboard to see what that does so that I can do right by pc people too?Sent from my iPhone

charles-plessy, 2024-10-17:
I have a JIS PC keyboard that I plugged on a Mac: Here are the results:
 - Muhenkan (無変換) -> `japanese_pc_nfer` 139
 - Henkan (変換) ->`japanese_pc_xfer` 138
 - Katakana Hiragana Romaji (カタカナ　ひらがな　ローマ字) -> `japanese_pc_katakana` 136

The Hankaku/Zenkaku Kanji (半角/全角・漢字) key did not produce japanese-specific code, just the `grave_accent_and_tilde` expected for its position (next to `1`). Also, the Eisuu (英数) command of the JIS PC keyboard, which should be shift-capsLock, just gave... shift-capsLock.  Maybe this is because the Mac does not care about these keys anyway.

I have not found yet a way to find this information on Windows (which I do not use) or Linux (my main platform). On Linux, `xev` gives keycodes that I do not know how to translate in scancodes.

charles-plessy, 2024-10-17:
Simplified output of `xev` on Linux with the same JIS keyboad, when selecting OADG 109A layout.

- Muhenkan (無変換)
```
    state 0x4000, keycode 102 (keysym 0xff22, Muhenkan), same_screen YES,
```
- Henkan (変換)
```
    state 0x4000, keycode 100 (keysym 0xff23, Henkan_Mode), same_screen YES,
```
 - Katakana Hiragana Romaji (カタカナ　ひらがな　ローマ字) 
```
    state 0x4000, keycode 101 (keysym 0xff27, Hiragana_Katakana), same_screen YES,
```
-  Hankaku/Zenkaku Kanji (半角/全角・漢字)
```
    state 0x4000, keycode 49 (keysym 0xff2a, Zenkaku_Hankaku), same_screen YES,
```
- Eisuu (英数)
```
    state 0x4000, keycode 66 (keysym 0xff30, Eisu_toggle), same_screen YES,
```
- Eisuu (英数)  (shift-capsLock)
```
    state 0x4001, keycode 50 (keysym 0xffe5, Caps_Lock), same_screen YES,
```
- ¥
```
    state 0x4000, keycode 132 (keysym 0xa5, yen), same_screen YES,
```
- \
```
    state 0x4000, keycode 97 (keysym 0x5c, backslash), same_screen YES,
```

charles-plessy, 2024-10-18:
OK, for Windows, from [Wikipedia](https://en.wikipedia.org/wiki/Scancode#cite_note-msusb13a-5) I found a link to a Microsoft document that specifies:
 - \ usage index 136
 - かな usage index 136
 - ¥ usage index 137
 - 変換 (XFER) usage index 138
 - 無変換 (NFER) usage index 139
I could not find Eisuu (英数) in this document…

closed 2024-11-11 by obra