keyboardio/Chrysalis #1358: Can't connect Model 100 to Chrysalis

person678, 2024-10-29:
My Model 100 keyboard now no longer connects to the Chrysalis software. The "Reading data from device" screen is displayed indefinitely. The firmware screen can be accessed, but I am also unable to flash firmware - never progresses past "Reading EEPROM".

I've tried:
Different PCs with different versions of Windows
Different USB cables
Different browsers (Chrome, Edge)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Chrysalis
2. Click on 'Connect'
3. Select COM Port

**Expected behavior**
For the interface for modifying the Model 100 firmware to be displayed.

**Screenshots**
Video attached.

**Desktop (please complete the following information):**
 - OS: Windows 10/Windows 11, possibly others

https://github.com/user-attachments/assets/07b46f95-297f-4b29-9167-d734d6851582

obra, 2024-11-11:
Can you go into the sidebar menu after the failure and pull a debugging log and attach it here?

person678, 2024-11-12:
@obra Just had a go at that but when I click "Create Bundle" nothing happens.

obra, 2024-11-18:
It might take a couple minutes, especially if Chrysalis has been open for a
while. If you open the browser's devtools, it should show a running log of
what it's trying to do in the console.

person678, 2024-11-20:
Tried waiting for a while but didn't seem to work unfortunately. 
Here is the console readout in google chromes dev settings, after clicking "connect" and selecting the keyboard:

```
[2024-11-20T10:01:54.946Z] [DEBUG] [c.resetDeviceState (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:205:352165)]: Resetting device state
main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:01:55.112Z] [DEBUG] [a.navigate (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:214:15194)]: navigating to location {args: Array(1)}
main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:01:55.139Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:214:499758)]: [object Object]
main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:01:55.139Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:214:499758)]: null
main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:01:58.618Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:214:499758)]: [object Object]
main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:01:58.618Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:214:499758)]: null
main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:01:59.966Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:214:499758)]: [object Object]
main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:01:59.966Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:214:499758)]: null
main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:02:02.061Z] [LOG] [async connectKeyboard (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:214:116389)]: The connected device: {usbProductId: 6, usbVendorId: 13462}
main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:02:02.061Z] [LOG] [async connectKeyboard (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:214:116389)]: Found a keyboard {info: {…}, usb: {…}, keyboard: {…}, components: {…}, flash: ƒ}
main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:02:02.061Z] [DEBUG] [c.resetDeviceState (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:205:352165)]: Resetting device state
main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:02:02.063Z] [LOG] [connectKeyboard (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:214:116429)]: about to nav to focus not detected   
main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:02:02.064Z] [INFO] [c._sendRequest (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:205:356231)]: Sending request {cmd: 'help', args: Array(0)}
main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:02:02.071Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:214:499758)]: [object Object]
main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:02:02.071Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:214:499758)]: [object Object]
main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:02:02.326Z] [DEBUG] [c._sendRequest (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:205:356658)]: response: keymap.custom
keymap.default
keymap.onlyCustom
help
device.reset
plugins
settings.defaultLayer
settings.valid?
settings.version
settings.crc
eeprom.contents
eeprom.free
eeprom.erase
version
keymap.layerNames
led_mode.default
led.brightness
spacecadet.mode
spacecadet.timeout
oneshot.timeout
oneshot.hold_timeout
oneshot.double_tap_timeout
oneshot.stickable_keys
oneshot.auto_mods
oneshot.auto_layers
escape_oneshot.cancel_key
macros.map
macros.trigger
mousekeys.scroll_interval
mousekeys.init_speed
mousekeys.base_speed
mousekeys.accel_duration
mousekeys.warp_grid_size
palette
colormap.map
idleleds.time_limit

main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:02:02.327Z] [DEBUG] [c.request (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:205:355573)]: enqueueing request: plugins {request: {…}}
main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:02:02.327Z] [INFO] [c._sendRequest (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:205:356231)]: Sending request {cmd: 'plugins', args: Array(0)}
main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:02:02.596Z] [DEBUG] [c._sendRequest (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:205:356658)]: response: EEPROMKeymap
LayerNames
DefaultLEDModeConfig
LEDBrightnessConfig
Qukeys
SpaceCadet
OneShot
OneShotConfig
EscapeOneShot
Macros
DynamicMacros
MouseKeys
MagicCombo
GeminiPR
ColormapEffect
PersistentIdleLEDs

main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:02:02.597Z] [DEBUG] [c.request (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:205:355573)]: enqueueing request: version {request: {…}}
main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:02:02.597Z] [INFO] [c._sendRequest (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:205:356231)]: Sending request {cmd: 'version', args: Array(0)}
main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:02:02.862Z] [DEBUG] [c._sendRequest (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:205:356658)]: response: 0.91.1+74 
[... cut ...] 535 65535 65535 18658 18666 18665 65535 49 2097 
main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:02:03.178Z] [DEBUG] [c.request (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:205:355573)]: enqueueing request: keymap.custom {request: {…}}
main.df320e2f48ad406ad30e.bundle.js:214 [2024-11-20T10:02:03.178Z] [INFO] [c._sendRequest (https://chrysalis.keyboard.io/js/main.df320e2f48ad406ad30e.bundle.js:205:356231)]: Sending request {cmd: 'keymap.custom', args: Array(0)}
```

obra, 2024-11-25:
This is looking like there's something in your customized key layouts that's causing Kaleidoscope to crash or freeze when Chrysalis asks for them.

And actually, that's one of the things that we put a lot of effort into for the 0.92 firmware series.  A factory reset and update will likely solve this for you, at the expense of your currently customized keyboard layouts. How much have you customized them? If there's a lot there, I can try to figure out a way to save them, but if replicating them would be straightforward or you have a backup, the easiest thing to do is to put your keyboard in recovery mode by holding in the prog key while connecting your Model 100 to the computer, visiting https://chrysalis.keyboard.io, and doing a factory reset.   When you first try to connect, the list of keyboards may be empty if you're in recovery mode. Hit cancel and you should see a "USB DFU" button show up. From there, you can factory reset and firmware update.
-jesse

person678, 2024-11-27:
Thank you! That worked perfectly, my customisations were not so extensive that I can't recreate them fairly quickly, so I just reset. Appreciate all your help.

closed 2024-11-27 by person678