keyboardio/Chrysalis #1359: Too many open issues and requests!

domalex, 2024-11-11:
I have decided to contribute on Kickstarter to your Preonic keyboard campaign. As Chrysalis is used to program all your keyboards I am deeply concerned that nobody seems to seriously following up on issues and requests reported by your customers. As of today (11.11.2024) there are 147 issues and 4 pull requests open, waiting to be solved! This situation requires serious improvement! The best keyboard isn't any good if the software to program it with isn't being developed timely.

obra, 2024-11-11:
This is a duplicate of the comment you posted on the preonic kickstarter. I'll reply to you there, likely later today.

closed 2024-11-11 by obra

domalex, 2024-11-12:
I posted it twice because the thread seems relevant for preonic campaign contributors as well as for current and future Chrysalis users.