keyboardio/Chrysalis #1372: Missed keystrokes when using secondary action

jacereda, 2024-12-20:
**Describe the bug**

My Atreus misses keystrokes in https://vice-emu.sourceforge.io/

**To Reproduce**  

1. Install Vice from the above URL.
2. Configure a key as Space when tapped and Layer Shift when held.
3. Launch `x64sc`.
4. Press the Space key several times, you should notice it fails to register the keypress quite often.

**Expected behavior**
Keystrokes should always be registered.

**Desktop (please complete the following information):** 

 - OS: NixOS and Windows
 - Chrysalis Version: Chrysalis 2024.1126.2237

**Additional context**

I suspect key up events happen too fast, maybe a setting to configure the time could help here?

obra, 2024-12-20:
Does it only happen with *space* and *layer shift*?

Does it matter how quickly you're tapping the key?

We don't have the full configuration options exposed in the focus protocol
Chrysalis talks, but if you're up for building your own firmware, there are
some more knobs there. But there's probably an opportunity for a more
sensible default behavior.

If you're comfortable with a text editor and a compiler, I'd love help
getting a reproducible test case together so we know what's going on and
when we've fixed it.

Tests are written as example arduino sketches with 'ktest' files that run
through a series of actions and expectations.
https://github.com/keyboardio/Kaleidoscope/tree/master/tests/plugins/Qukeys/basic
is an example that shows off some testing of a regular+layer shift key.

You run the tests from a git checkout of kaleidoscope by running 'make
simulator-tests'

Let me know what I can answer to help out.

jacereda, 2024-12-22:
I only tried Layer Shift, but it also fails with Enter and Backspace.

I also have a Chocofi with pretty much the same configuration and also suffers from the same problem in Vice. I've just reflashed qmk in the Chocofi with `#define TAP_CODE_DELAY 40` and it fixed the issue. Also tried with a 20 ms delay, but it was still missing keystrokes from time to time. 

Is there any equivalent to that define in Kaleidoscope?

jacereda, 2024-12-22:
Same issue in qmk: https://github.com/qmk/qmk_firmware/issues/24741