keyboardio/Chrysalis #1374: Failing to connect to Model 01

durka, 2025-03-04:
**Describe the bug**
I can't get Chrysalis to connect to my Model 01.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://chrysalis.keyboard.io in Chrome (I also tried using `yarn start` and `localhost:8080`)
2. Click on CONNECT
3. Select  `Model 01 (ttyACM0)` from the menu
4. Wait for connect button to stop spinning. Nothing happens after that.
5. Repeat steps 2-3.
6. Receive error: `I couldn't connect to the keyboard. It may already be connected in another application or browser window. The error I got was: InvalidStateError: Failed to execute 'open' on 'SerialPort': The port is already open.`

**Expected behavior**
It should connect to the keyboard.

**Screenshots**
I can see this in the browser console:

[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 24.04 with Chrome 133.0.6943.141
 - Chrysalis Version: 2025.0224.2100

**Additional context**
I added the udev rules suggested on the error page, but it didn't help.

obra, 2025-03-04:
The "port is already open" suggests that there's something that's
connecting to the serial port, blocking Chrysalis' connection. If that's
chrysalis itself, a browser force-refresh ought to do it. If it's something
else like Cura or Modem Manager, that other tool needs to be stopped first.

durka, 2025-03-05:
I see, it was connecting, but not really indicating that and I'm supposed to go to the hamburger menu to navigate away from the connection page

closed 2025-03-05 by durka

durka, 2025-03-05:
It turns out the installed firmware was too old so it was sort of connecting but not really. I had to do this: https://github.com/keyboardio/Chrysalis/issues/1335#issuecomment-2344762515