keyboardio/Chrysalis #1387: Chrysalis doesn't see model 101

e-max, 2025-05-27:
Hey guys!

**Describe the bug**

Chrysalis doesn't recognize Model 101

I've tried the current web version, pressed the "Connect" button but it shows "No compatible devices found" window. 

That's what I see in a dev console
```
[2025-05-27T20:54:41.897Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:599625)]: [object Object]
main.0d26ab7476504d629676.bundle.js:214 [2025-05-27T20:54:41.897Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:599625)]: null
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000110e-0000-1000-8000-00805f9b34fb, name='Dixie Dewdrop'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000111e-0000-1000-8000-00805f9b34fb, name='Dixie Dewdrop'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000110c-0000-1000-8000-00805f9b34fb, name='Dixie Dewdrop'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000110b-0000-1000-8000-00805f9b34fb, name='Dixie Dewdrop'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=00001108-0000-1000-8000-00805f9b34fb, name='Dixie Dewdrop'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000110e-0000-1000-8000-00805f9b34fb, name='Jabra Elite 85h'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000111e-0000-1000-8000-00805f9b34fb, name='Jabra Elite 85h'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000110c-0000-1000-8000-00805f9b34fb, name='Jabra Elite 85h'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000110b-0000-1000-8000-00805f9b34fb, name='Jabra Elite 85h'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=00001108-0000-1000-8000-00805f9b34fb, name='Jabra Elite 85h'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000111e-0000-1000-8000-00805f9b34fb, name='Soundcore Space A40'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000110e-0000-1000-8000-00805f9b34fb, name='Soundcore Space A40'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000110c-0000-1000-8000-00805f9b34fb, name='Soundcore Space A40'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=00007034-0000-1000-8000-00805f9b34fb, name='Soundcore Space A40'
chrysalis.keyboard.io/:1 Chooser dialog is not displaying a port blocked by the Serial blocklist: bluetoothServiceClassId=0000110b-0000-1000-8000-00805f9b34fb, name='Soundcore Space A40'
main.0d26ab7476504d629676.bundle.js:214 [2025-05-27T20:55:10.430Z] [ERROR] [https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:172325]: I couldn't connect to your keyboard's serial port. That might be because another program or browser window is already connected. NotFoundError: Failed to execute 'requestPort' on 'Serial': No port selected by the user.
main.0d26ab7476504d629676.bundle.js:214 [2025-05-27T20:55:10.430Z] [LOG] [async U (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:116760)]: No serialport selected
main.0d26ab7476504d629676.bundle.js:214 [2025-05-27T20:55:10.430Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:599625)]: [object Object]
main.0d26ab7476504d629676.bundle.js:214 [2025-05-27T20:55:10.431Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:599625)]: null
```

This is the debug bundle:

```
{
  "logs": [
    "[2025-05-27T20:32:38.908Z] [DEBUG] [c.resetDeviceState (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:205:400333)]: Resetting device state",
    {"message": "[2025-05-27T20:32:39.032Z] [DEBUG] [https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:15519]: navigating to location", "data": {"args": ["/keyboard-select"]}},
    "[2025-05-27T20:32:40.770Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:599625)]: [object Object]",
    "[2025-05-27T20:32:40.771Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:599625)]: null",
    "[2025-05-27T20:32:49.727Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:599625)]: [object Object]",
    "[2025-05-27T20:32:49.727Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:599625)]: null",
    {"message": "[2025-05-27T20:33:02.760Z] [ERROR] [https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:172325]: I couldn't connect to your keyboard's serial port. That might be because another program or browser window is already connected.", "data": {}},
    "[2025-05-27T20:33:02.760Z] [LOG] [async U (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:116760)]: No serialport selected",
    "[2025-05-27T20:33:02.761Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:599625)]: [object Object]",
    "[2025-05-27T20:33:02.959Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:599625)]: null",
    "[2025-05-27T20:33:57.305Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:599625)]: [object Object]",
    "[2025-05-27T20:33:57.305Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:599625)]: null",
    {"message": "[2025-05-27T20:34:30.598Z] [ERROR] [https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:172325]: I couldn't connect to your keyboard's serial port. That might be because another program or browser window is already connected.", "data": {}},
    "[2025-05-27T20:34:30.599Z] [LOG] [async U (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:116760)]: No serialport selected",
    "[2025-05-27T20:34:30.600Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:599625)]: [object Object]",
    "[2025-05-27T20:34:30.601Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:599625)]: null",
    "[2025-05-27T20:34:33.036Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:599625)]: [object Object]",
    "[2025-05-27T20:34:33.036Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:599625)]: null",
    {"message": "[2025-05-27T20:34:41.339Z] [ERROR] [https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:172325]: I couldn't connect to your keyboard's serial port. That might be because another program or browser window is already connected.", "data": {}},
    "[2025-05-27T20:34:41.340Z] [LOG] [async U (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:116760)]: No serialport selected",
    "[2025-05-27T20:34:41.341Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:599625)]: [object Object]",
    "[2025-05-27T20:34:41.341Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:599625)]: null",
    "[2025-05-27T20:34:51.793Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:599625)]: [object Object]",
    "[2025-05-27T20:34:51.793Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:599625)]: null",
    {"message": "[2025-05-27T20:35:02.562Z] [ERROR] [push.60290.a.connectToDfuUsbPort (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:171805)]: Failed to open usb port", "data": {}},
    "[2025-05-27T20:35:02.564Z] [LOG] [Sn (https://chrysalis.keyboard.io/js/main.0d26ab7476504d629676.bundle.js:214:599625)]: [object Object]",
[... cut ...]
**Desktop (please complete the following information):**
 - OS: Windows 11
 - Chrysalis Version: 2025.0507.1815

**Additional context**

What else did I try
* Plug in the keyboard with the Prog button pressed. I see red light but Chrysalis still doesn't recognize it
* Download an old Windows desktop version of Chrysalis - the same issue, it doesn't recognize the keyboard. 

Maybe the reason is that the firmware is old?  The keyboard is part of the original Kickstarter order, it worked great but I had to put it off for some time.

e-max, 2025-05-28:
For some reason, Chrysalis couldn't detect the keyboard when it was plugged in via a USB hub. However, when I connected it directly to a USB port on the PC, it was able to detect it. 
But after I've updated the firmware,  it works even through the USB hub. 
All is good now. Thanks! Let me close the PR

closed 2025-05-28 by e-max

obra, 2025-05-29:
Awesome. I'm so happy to hear that it's behaving now.
-jesse