keyboardio/Chrysalis #1402: Paste layer crashes with Keyboardio Preonic

aspiers, 2025-08-13:
**Describe the bug**
It crashes every single time I try to paste a layer when configuring my Preonic.

**To Reproduce**
Steps to reproduce the behavior:
1. Load the site
2. Connect a Keyboardio Preonic
3. Click "Copy layer"
4. Select another layer
5. Click "Paste layer"

**Expected behavior**
It pastes the layer.

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: openSUSE Tumbleweed
 - Chrysalis Version: 2025.0809.2112

**Additional context**
Paste layer seems to work fine with the Keyboardio Atreus, so maybe it's specific to the Preonic.

obra, 2025-08-13:
I'm trying to repro this here and having trouble. Can you attach an export/backup of your current layers so I can clone them onto a device to test with?

aspiers, 2025-08-13:
Sure, here you go:

[Chrysalis_Keyboardio-Preonic_layout_2025-08-13_18-41-01.json](https://github.com/user-attachments/files/21759789/Chrysalis_Keyboardio-Preonic_layout_2025-08-13_18-41-01.json)