keyboardio/Chrysalis #1407: Reset to factory defaults broken on Keyboardio Atreus

looi, 2025-09-29:
**Describe the bug**
In firmware 0.94.0+132, reset to factory defaults is broken on Keyboardio Atreus.

**To Reproduce**
Steps to reproduce the behavior:
1. Reset to factory defaults (together with flashing firmware or not)
2. Default layout is not seen

After downgrading to firmware 0.92.6+116, the reset works as expected.

obra, 2025-09-29:
Can you provide a bit more information on how you're doing this and exactly
what you're seeing, along with the debugging bundle from the sidebar?

Are you using factory reset from the flashing screen? If so, you *can't* do
a factory reset without flashing the firmware.

Were you doing the update from bootloader mode or from keyboard mode?

looi, 2025-09-29:
I did an update from bootloader mode to 0.94.0+132, checking the box to reset to default settings. The firmware then successfully updated but the settings were not reset. I then tried to reset the settings again from keyboard mode, which did not work. Then I went to bootloader mode again and flashed the same firmware version. Also no reset. Then I went to bootloader mode again and flashed 0.92.6+116. Now the settings were reset.

I no longer have this device as I was doing this reset to prepare the keyboard for its new owner.

MinceMan, 2025-10-02:
I just updated to `0.94.0+132` as well.

If I try to update key binds through the GUI and apply the keybinds to the Atreus I get very weird random binds across all of my layers. There isn't even a functioning enter.

If I flash a backup, it works correctly.

It also works correctly if I
1. make GUI key binding changes
2. backup changes
3. restore changes
4. apply the backup.

eritain, 2026-01-01:
Further details, in descending order of probable relevance:

The weird random keybinds appeared as soon as I updated to 0.94.0+132. From that state, I was able to set keycodes in Chrysalis and save them to the Atreus, or to load a keymap from a JSON backup file and save it to the Atreus. 

Chrysalis appeared to set "sticky" modifiers, reported them correctly as one-shot in a JSON backup file, and was able to save them to the Atreus and read them back from it. They just didn't "stick" in actual use.

Attempting to install the latest firmware from a file of my own, as if a custom install, did not fix the problem.

Chrysalis's usual action on being asked to do a factory reset was to claim success while actually failing, but on a few occasions I got it to fail in some other way and/or recognize an error condition. These may or may not have anything to do with the firmware problem that caused the random binds and broke the factory reset routine:

* Chrysalis 0.13.3 on macOS, keyboard mode, factory reset got as far as the "hold Escape and click Continue" dialog, and then the app hung: Clicking "Continue" didn't advance it in the process, clicking "Cancel" didn't back out. However, starting from bootloader mode (without holding Escape again at that step, IIRC) did get the usual false success. However, that version is known to have been kind of crappy at communicating with the keyboard.
* Chrysalis on the web, imitating [the manual fix for the Model 100 firmware corruption problem](https://community.keyboard.io/t/model-100-firmware-update-to-fix-corruption-issues/5553) got an "Error flashing the firmware" message immediately after hitting "Update anyway" the second time around. However, that's a deliberately weird procedure intended to be used while upgrading to non-buggy firmware.
* Chrysalis 0.13.3 on Windows (10? Not sure), bootloader mode, factory reset once got a "keyboard disconnected" message. However, another time it got the usual false success, and it also got the false success when using keyboard mode. And the OS was not fully up to date, and the machine was running incredibly slow for unknown reasons.
* Chrysalis on the web, bootloader mode, on the 'you can try clicking the "Reconnect" button' dialog, actually clicking the "Reconnect" button got an error "TypeError: e.setLayerSize is not a function." But "Reconnect" isn't really intended for when the keyboard is in bootloader mode anyway.

eloyesp, 2026-02-14:
i've just compiled the default firmware for atreus using the master branch on kaleidoscope and it happens the same thing, the eeprom memory is still there and some keys don't work as expected (like the shift on the secondary action)

hairboat, 2026-11-16:
Thank you @eritain for the detail here. Confirming I experienced the same problem in Nov 2026. Not being confident in how to downgrade the firmware I simply rebuilt my layers by hand 🙃 

I had the same issue with secondary action shift as @eloyesp - disabling Space Cadet in settings solved it, which only works if you didn't want to use Space Cadet, which I didn't, fortunately!